Xilinx/Vitis_Accel_Examples #1: hello_world/Makefile

gkrislara, 2020-01-03:
I found a few issues while running commands using make 

1)make build is not working
2)make all reports error for HOST_ARCH= aarch64
 
fatal error: CL/cl2.hpp: No such file or directory
 #include <CL/cl2.hpp>
          ^~~~~~~~~~~~

vishnuchebrolu, 2020-01-03:
Hi @gkrislara ,

Please see my in-line response
 1. make build is not working : 
     Setting the PLATFORM_REPO_PATHS to installation folder of platform files can enable makefiles in this library to use DEVICE variable as a pattern. Otherwise, full path to .xpfm file needs to be provided via DEVICE variable. (check "make help" command regarding the Makefile usage).
2. make all reports error for HOST_ARCH= aarch64
     Did you sourced the Xilinx XRT before running the design?
     You need to provide SYSROOT if you running for SOC Shells.

You can post the same query in xilinx forums for Vitis related questions. Please find below link for Xilinx forums : 
https://forums.xilinx.com/t5/Vitis-SDAccel-and-SDSoC/bd-p/tools_v

gkrislara, 2020-01-03:
Thanks for the reply
1) I have already set PLATFORM_REPO_PATHS to the platform installation directory
2) I have already initialised XRT and provided sysroot

The complier defined in makefile includes library from common/include (available in this repository)   where xcl2.hpp and crt are present

heeran-xilinx, 2020-04-29:
closing this issue as fixed. if issue still there, please reopen.

closed 2020-04-29 by heeran-xilinx