Xilinx/Vitis_Accel_Examples #5: Issue in SLR - assign

mahinlma, 2020-02-26:
In this https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/sys_opt/slr_assign
after generating binary, I just verified binary with vitis_analyser. Both the kernels assigned in bank1.

`Expected response - vmult to  SLR0, vadd to SLR1`

Please let me know, what will be the issue.
Thanks

vishnuchebrolu, 2020-02-26:
Hi @mahinlma ,
Currently if you check the vmult_vadd.ini file inside slr_assign/, we have both vmult and vadd mapped to SLR1.
We need to update the readme where both vmult and vadd map to SLR1.

mahinlma, 2020-02-26:
@vishnuchebrolu , I changed vmult_vadd.ini file before running `make all TARGET=hw`
(vmult to SLR0, vadd to SLR1)

heeran-xilinx, 2020-04-29:
Hi @mahinlma ,
Please open a discussion a topic on Xilinx Forum:
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v

You can use this git issues if we are facing any issue of using current setup.

closed 2020-04-29 by heeran-xilinx