Xilinx/Vitis_Accel_Examples #6: PLRAM Example shows error

aeeshms, 2020-03-16:
The plram example shows error as:
[XRT] ERROR: context is nullptr
[XRT] ERROR: context is nullptr

mullt.ini has the connectivity but still its not able to differentiate banks
The flag XCL_MEM_PLRAM_BANK3 is not applicable to plram.
What flay can be used used for PLRAM

vishnuchebrolu, 2020-03-17:
Hi @aeeshms ,
Tried running the plram design and it's working fine without any error.

Can you tell me what exactly is the change you are looking in mmult.ini. If you want to use different PLRAM you can pick from following(PLRAM[0], PLRAM[1] and PLRAM[2]) similarly if different bank is required(DDR[0], DDR[1], DDR[2] and DDR[3]). These PLRAM and bank information is for U200 platform. 
Based on platform the resource changes(PLRAM, DDR).

heeran-xilinx, 2020-03-24:
Closing as no response got from filer for last query. Feel free to re-open if this issue still there.

closed 2020-03-24 by heeran-xilinx