Xilinx/Vitis_Accel_Examples #8: Error when creating example project of RTL Kernel Wizard IP

Hecmay, 2020-04-12:
I am following the  `rtl_streaming_k2k_mm` example to package RTL kernels into XO files, and make them callable from host. The TCL script for generating XO files (i.e.`src/gen_xo_myadder1.tcl`) works fine with the default kernel type `RTL`, but failed with the type `Block Design`. The error message is shown as below.

```shell
ERROR: [xilinx.com:ip:rtl_kernel_wizard:1.0-100] Failed to execute xsct to create and build software: "child process exited abnormally", please check rtl_streaming_k2k_mm/myadder2_ex/myadder2_ex.sdk/xsct.log for more details.
ERROR: [open_example_project] Open Example Project failed: Error encountered while sourcing custom IP example design script.
ERROR: [Common 17-39] 'send_msg_id' failed due to earlier errors.
```

Error message from the log file:

```shell
Starting vitis. This could take few seconds...WARNING: Using incubator modules: jdk.incubator.httpclient
done
INFO: [Hsi 55-2053] elapsed time for repository (/work/shared/common/Xilinx/Vitis/2019.2/data/embeddedsw) loading 2 seconds
Opening the hardware design, this may take few seconds.
/work/shared/common/Xilinx/Vitis/2019.2/gnu/microblaze/lin
error deleting "./src/arm/common": file already exists
ERROR: [Hsi 55-1545] Problem running tcl command ::sw_standalone_v7_1::generate : error deleting "./src/arm/common": file already exists
    while executing
```

From the document, the RTL Kernel Wizard will generate a MicroBlaze IP for the example project with kernel type "Block design". What is the cause this issue? Is there any tutorial on how to create RTL kernel for Vitis using custom BD design?

vishnuchebrolu, 2020-04-13:
Hi @Hecmay ,
Can you provide more details regarding the setup you have used.
1. Using Command line flow or GUI flow?
2. Facing the above issue for Emulation flow(hw_emu) or Hardware run?
3. Using 2019.2 release?

We tried locally running the design(command line flow) and it's passing for both Emulation and board runs.

Hecmay, 2020-04-13:
@vishnuchebrolu Thanks for your reply.

I am using command line flow in Vitis 2019.2 (I also tried the GUI flow. It works fine) the issue occurred before the emulation for actual hardware runs. 

Here is the TCL file I used (It's almost exactly the same with the example, apart from the kernel type). It returns the error when executing `open_example_projec`.

```tcl
create_ip -name rtl_kernel_wizard -vendor xilinx.com -library ip -version 1.0 -module_name rtl_kernel_wizard_0

set_property -dict [list CONFIG.KERNEL_NAME {rtl_kernel_wizard_0} CONFIG.KERNEL_TYPE {Block_Design} CONFIG.NUM_RESETS {1} CONFIG.NUM_AXIS {1}] [get_ips rtl_kernel_wizard_0]

generate_target {instantiation_template} [get_files rtl_kernel_wizard_0.xci]
set_property generate_synth_checkpoint false [get_files rtl_kernel_wizard_0.xci]
generate_target all [get_files rtl_kernel_wizard_0.xci]

# the error occurs in their line
open_example_project [get_ips rtl_kernel_wizard_0]
```

cpgithub2020, 2020-04-14:
Hi @Hecmay,
Please try following:
1) use "open_example_project -force -in_process -dir [pwd] [get_ips rtl_kernel_wizard_0]" 
2) run your script in a new working directory
3) make sure there are no leftover processes related to your previous runs from your script

heeran-xilinx, 2020-04-29:
closing this issue as solution is provided in comments. if issue still exist, please re-open it.

closed 2020-04-29 by heeran-xilinx