Xilinx/Vitis_Accel_Examples #9: example "hello_world"

rbhuang-nokia, 2020-06-12:
hi

Has anyone gone through the "hello_world" example under "VITIS_Accel_examples"?Will you share your steps?

../src/host.hpp  invokes:
#include "xcl2.hpp"
#include <algorithm>
#include <vector>
#define DATA_SIZE 4096

and xcl2.hpp invokes:
#include <CL/cl2.hpp>
#include <CL/cl_ext_xilinx.h>
I can not find cl2.hpp ,cl_ext_xilinx.h.

vishnuchebrolu, 2020-06-12:
Hi @rbhuang-nokia ,

Looks like the OpenCL path is not able to picked by the design.
I hope you have sourced the Vitis and XRT before running the design.
1. source <path to vitis setup.sh>
2. source <path to xrt setup.sh>

Can you provide more details on the Vitis version you are using and OS details.

Thanks,
Vishnu

rbhuang-nokia, 2020-06-12:
hi  vishnu

I am using centos7.7(since xilinx recommends centos7.4/5/6) and vivado/vitis 2019.2.

I added following at the end of  .cshrc:
....................
source */Vitis/2019.2/settings64.csh
source /opt/xilinx/xrt/setup.csh
setenv PLATFORM_REPO_PATHS /opt/xilinx/platforms

I list (env) some variable as following:
................
XILINX_VITIS=*/Vitis/2019.2
XILINX_VITIS_HLS=*/Vitis/2019.2
vitis_PATH=*/Vitis/2019.2/bin
vitis_hls_PATH=*/Vitis/2019.2/bin
XILINX_XRT=/opt/xilinx/xrt
PYTHONPATH=/opt/xilinx/xrt/python
PLATFORM_REPO_PATHS=/opt/xilinx/platforms

vishnuchebrolu, 2020-06-12:
Hi @rbhuang-nokia ,

Xilinx recommends Centos version of 7.4/7.5/7.6 But looks like you are using centos7.7 version. I think It's not recommended to use different versions other than supported.

Thanks,
Vishnu

pooyaww, 2020-06-12:
Check that OpenCL headers are already installed in your system, otherwise you can't compile the host code.

rbhuang-nokia, 2020-06-12:
:>rpm -qa | grep opencl
opencl-headers-2.2-1.20180306gite986688.el7.noarch
[root@fpga ~]# 

Does this OK?

pooyaww, 2020-06-12:
Seems fine from package manager point of view. But eventually you should have them in /usr/include/CL/ . besides, cl_ext_xilinx.h is found via XRT you can't find it in that path. Also, check for xilinx.icd file in /etc/OpenCL/vendors/ . If all are fine print your env variables and see if they are set accordingly.

rbhuang-nokia, 2020-06-15:
Hi all

I have tranfered to centos7.6 and the results are the same as centos7.7.

I check that cl2.hpp is in /usr/include/CL/ , cl_ext_xilinx.h is not there. Also, xilinx.icd file is in /etc/OpenCL/vendors/.

To simple,I changed to absolute path as following:
…………..
#include "/home/rbhuang/Vitis_Accel_Examples-master/common/includes/xcl2/xcl2.hpp"
#include <algorithm>
…………………………………………..  in host.cpp

………………….
#include </usr/include/CL/cl2.hpp>
#include </usr/include/CL/cl_ext_xilinx.h>
…………………………………………..in xcl2.hpp

And with command:>g++ host.cpp ,error listed:

[rbhuang@fpga1 src]$ g++ host.cpp
In file included from /home/rbhuang/Vitis_Accel_Examples-master/common/includes/xcl2/xcl2.hpp:53:0,
                 from host.cpp:36:
/usr/include/CL/cl2.hpp:493:2: error: #error Visual studio 2013 or another C++11-supporting compiler required
#error Visual studio 2013 or another C++11-supporting compiler required
  ^
In file included from /usr/include/c++/4.8.2/mutex:35:0,
                 from /usr/include/CL/cl2.hpp:542,
                 from /home/rbhuang/Vitis_Accel_Examples-master/common/includes/xcl2/xcl2.hpp:53,
                 from host.cpp:36:
/usr/include/c++/4.8.2/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
#error This file requires compiler and library support for the \
  ^
In file included from host.cpp:36:0:
/home/rbhuang/Vitis_Accel_Examples-master/common/includes/xcl2/xcl2.hpp:54:43: fatal error: /usr/include/CL/cl_ext_xilinx.h: No such file or directory

BR

vishnuchebrolu, 2020-06-15:
Hi @rbhuang-nokia ,
Can you search for cl_ext_xilinx.h file under XILINX_XRT. You should be able to find cl_ext_xilinx.h under XILINX_XRT.

Thanks,
Vishnu

pooyaww, 2020-06-15:
You cannot compile the host code without configuring the g++ flags. Try using *cmake* flow using the *CMakeLists.txt* available in the *cmake_build* folder.

rbhuang-nokia, 2020-06-16:
Hi

I　use the cmake under vitis 2019.2 in GUI and generated some files in XCLBIN as following.

Then,what will be next?

[cid:image001.png@01D643BD.E9F5AC40]

rbhuang-nokia, 2020-06-16:
Hi puya amiri

I am trying compile the host.cpp for hello_world with following command:
:>g++ -I$XILINX_XRT/include -I$XILINX_VIVADO/include -std=c++11 -c -I './src' -o 'host.o' ./src/host.cpp

and got many deprecated unit.Is there any  version mismatched?

[cid:image002.png@01D643F6.1C410F90]

BR

rbhuang-nokia, 2020-06-18:
I read of a solution from internet to resolve the deprecated problem.Adding "#define CL_USE_DEPRECATED_OPENCL_1_1_APIS" before
"#include </usr/include/CL/cl2.hpp>" in xcl2.hpp file.

Now ,compiling the host.cpp to get host.o is ok,but while generating host.exe ,errors accured.
./src/host.o: In function `main':
host.cpp:(.text+0x2a6): undefined reference to `xcl::get_xil_devices()'
host.cpp:(.text+0x2bf): undefined reference to `xcl::read_binary_file(std::string const&)'
collect2: error: ld returned 1 exit status

After upgrading vitis and xrt to 2020.1,these errors are still there.

vishnuchebrolu, 2020-09-14:
@rbhuang-nokia  were you able to solve the issue? If not try using latest g++ version and check if it works.

heeran-xilinx, 2020-10-06:
Closing as no response from reporter.  Please re-open if you still facing this issue and provide more info.

closed 2020-10-06 by heeran-xilinx