Xilinx/Vitis_Accel_Examples #11: How can I increase the width of axi control address in the shell?

saitej25, 2020-06-17:
https://github.com/Xilinx/Vitis_Accel_Examples/blob/9231c73b3849b0ef7fca756086db3e3912f51dbb/rtl_kernels/rtl_vadd/src/hdl/krnl_vadd_rtl.v#L45

vishnuchebrolu, 2020-06-22:
Hi @saitej25,
I think you can just change the parameter and it should increase.   You would need to update the control registers to match: 
https://github.com/Xilinx/Vitis_Accel_Examples/blob/9231c73b3849b0ef7fca756086db3e3912f51dbb/rtl_kernels/rtl_vadd/src/hdl/krnl_vadd_rtl_control_s_axi.v

I don’t see any dependencies elsewhere, but do note if they want to expand the range beyond 4kB (0x1000), you need to also change the range attribute in the kernel.xml for the s_axi_control port.

Thanks,
Vishnu

saitej25, 2020-06-22:
I tried it, but I'm getting errors "ERROR: [HW-EM 21] xclRegRW - invalid CU offset: 81920"

vishnuchebrolu, 2020-06-24:
Hi @saitej25,
I hope you have not update the kernel.xml.  I recommend you to follow the documentation: https://www.xilinx.com/html_docs/xilinx2020_1/vitis_doc/devrtlkernel.html#qnk1504034323350 

Thanks,
Vishnu

heeran-xilinx, 2020-07-27:
closing this issue assuming provided link was useful. Please re-open if it is still an issue and provide more steps to reproduce.

closed 2020-07-27 by heeran-xilinx