Xilinx/Vitis_Accel_Examples #13: hello_world and nimbix/jarvice

eejlny, 2020-07-03:
Hello,

I am trying to compile the hello_world example in Nimbix following instructions https://alveo.readthedocs.io/en/latest/vitis/.

To my surprise the compilation with 

TMP_WORK_DIR=$(mktemp -d) && make -C /data/Vitis_Accel_Examples/hello_world all BUILD_DIR=${TMP_WORK_DIR} TARGET=hw DEVICE=xilinx_u200_xdma_201830_2 && cp ${TMP_WORK_DIR}/*.xclbin /data/Vitis_Accel_Examples/hello_world

I get :

INFO: [v++ 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints. 
ERROR: [v++ 200-61] /data/Vitis_Accel_Examples/hello_world/src/vadd.cpp:100: unsupported memory access on variable 'in1' which is (or contains) an array with unknown size at compile time. 
ERROR: [v++ 200-70] Synthesizability check failed. 
ERROR: [v++ 60-300] Failed to build kernel(ip) vadd, see log for details: /data/Vitis_Accel_Examples/hello_world/_x.hw.xilinx_u200_xdma_201830_2/vadd/vadd/vivado_hls.log 
ERROR: [v++ 60-599] Kernel compilation failed to complete 
ERROR: [v++ 60-592] Failed to finish compilation 

Is there a problem with this simple example  in Nimbix?

I tried also with the kmeans example but it also failed with a different error:

ERROR: [CF2BD 82-34] cf_xsd: Failed to generate design file: dr.bd.tcl 
ERROR: [SYSTEM_LINK 82-36] [07:24:19] cf2bd failed 
Time (s): cpu = 00:00:02 ; elapsed = 00:00:02 . Memory (MB): peak = 180.418 ; gain = 0.000 ; free physical = 63829 ; free virtual = 196380 
ERROR: [SYSTEM_LINK 82-61] Error generating design file dr.bd.tcl 
ERROR: [SYSTEM_LINK 82-80] Unable to create top-level block diagram 
INFO: [v++ 60-1442] [07:24:19] Run run_link: Step system_link: Failed 
Time (s): cpu = 00:00:11 ; elapsed = 00:00:13 . Memory (MB): peak = 561.820 ; gain = 0.000 ; free physical = 63848 ; free virtual = 196399 
ERROR: [v++ 60-661] v++ link run 'run_link' failed 
ERROR: [v++ 60-626] Kernel link failed to complete 
ERROR: [v++ 60-703] Failed to finish linking 
Makefile:115: recipe for target '/tmp/tmp.FzMt2iMNkW/krnl_kmeans.xclbin' failed 
make: *** [/tmp/tmp.FzMt2iMNkW/krnl_kmeans.xclbin] Error 1 
make: Leaving directory '/data/Vitis_Accel_Examples/demo/kmeans'

mariodruiz, 2020-07-14:
I think the issue may be related to the Vitis version. In 2019.2, Vivado HLS was used by default. In Vitis 2020.1, Vitis-HLS is used by default. Pragmas have change from one version to another, the master is for Vitis 2020.1. If you are using Vitis 2019.2 checkout to the 2019.2 tag and try again.

eejlny, 2020-07-15:
OK, thanks,

That must be it. I missed that part.

heeran-xilinx, 2020-07-27:
closing this as issue is resolved as per last comment.

closed 2020-07-27 by heeran-xilinx