Xilinx/Vitis_Accel_Examples #15: Issues booting sw_emu tftp

jrjbertram, 2020-07-16:
Hi there... 

I've been trying to get the hello world example to work in vitis 2019 the past day or so.  I've ended up finding that these instructions here had a lot of the necessary background that helped getting started.  (I'd like to suggest that something with this level of detail gets pulled into this readme for xilinx noobs like myself.)

https://www.hackster.io/news/vivado-vitis-2019-2-install-on-ubuntu-18-04-lts-93242be6c9eb

One issue I'm stumped on is how to get my xilinx project to put its output into my hosts /tftpboot directory.   I have a tftpserver configured on my ubuntu linux host, and I have a /tftpboot which I believe is set with correct permissions.  

I found here on page 62  https://www.xilinx.com/support/documentation/sw_manuals/xilinx2019_2/ug1144-petalinux-tools-reference-guide.pdf that there's a petalinux-config command that should allow me to set a Copy final images to tftpboot checkbox, but when I run petalinux-config within the hello_world example's folder on my host, I get an error that I am not inside a PetaLinux project.

I tried running petalinux-config also from the application project that I created from the hello_world template from within the 2019.2 IDE.  Any ideas?

Thanks,
- J.

jrjbertram, 2020-07-23:
~ crickets ~

heeran-xilinx, 2020-07-27:
Hi @jrjbertram ,
Thanks for your query. We are looking into this and discussing the same within internal team. We will get back to you soon.

jrjbertram, 2020-07-27:
That's great to know, thanks.  Also, I've opened a somewhat related post on
the Xilinx support forums:

https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/vitis-hello-world-tutorials-on-zcu102-sw-emu/td-p/1131063

It's also focused on trying to get the sample apps running in sw_emu mode.
I think it's probably related to the tftpboot errors I was describing in
this post / issue.

- Josh.

heeran-xilinx, 2020-08-17:
closing this issue as there is similar issue getting discussed in Xilinx Forum.
Please re-open if you see this is a different issue.

closed 2020-08-17 by heeran-xilinx