Xilinx/Vitis_Accel_Examples #16: Cpp_kernels/Systolic_Array: large runtime and excessive memory usage problem

kaanolgu, 2020-07-19:
Hello,
I was trying to tune the size of the systolic array example (Default 16x16 matrix multiplication) I changed the DATA_SIZE and MAX_SIZE to 256 in host.cpp file and I changed the MAX_SIZE in mmult.cpp to 256 too. The sw_emu gave me correct results and showed that TEST PASSED but when I tried to the hw part it gave me the following error:

`===>The following messages were generated while  performing high-level synthesis for kernel: mmult Log file: /hdd2/users/sq19469/projects/Vitis_Accel_Examples/cpp_kernels/systolic_array256/_x.hw.xilinx_u200_xdma_201830_2/mmult/mmult/vivado_hls.log :
INFO: [v++ 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
ERROR: [v++ 203-504] Stop unrolling loop 'systolic2' (/hdd2/users/sq19469/projects/Vitis_Accel_Examples/cpp_kernels/systolic_array256/src/mmult.cpp:183) in function 'mmult' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [v++ 200-70] Pre-synthesis failed.
ERROR: [v++ 60-300] Failed to build kernel(ip) mmult, see log for details: /hdd2/users/sq19469/projects/Vitis_Accel_Examples/cpp_kernels/systolic_array256/_x.hw.xilinx_u200_xdma_201830_2/mmult/mmult/vivado_hls.log
ERROR: [v++ 60-599] Kernel compilation failed to complete
ERROR: [v++ 60-592] Failed to finish compilation
Makefile:97: recipe for target '_x.hw.xilinx_u200_xdma_201830_2/mmult.xo' failed
make: *** [_x.hw.xilinx_u200_xdma_201830_2/mmult.xo] Error 1`

So I changed the MAX_SIZE of mmult.cpp to 128,64,32,16 and in all cases the calculation results of FPGA didn't match to the CPU results. What could be the ideal solution for this issue ? The line 183 in mmult corresponds to the systolic2: part

Thank you in advance

virata-xilinx, 2020-07-20:
Hi,

I tried running the example cpp_kernels/systolic_array for hw_emu for 16, 32 and 64 DATA_SIZE cases (By setting the sizes just like you did for 256). I got TEST PASSED for all with the following log -

TEST PASSED
INFO::[ Vitis-EM 22 ] [Time elapsed: 12 minute(s) 4 seconds, Emulation time: 1.02774 ms]
Data transfer between kernel(s) and global memory(s)
mmult_1:m_axi_gmem-DDR[1]          RD = 32.000 KB              WR = 16.000 KB       

INFO: [HW-EM 06-0] Waiting for the simulator process to exit
INFO: [HW-EM 06-1] All the simulator processes exited successfully

Can you please tell me what were the steps followed by you? Also which VITIS version did you source?

virata-xilinx, 2020-07-20:
Hi,

I have tried for the master branch and sourced VITIS 2020.1 version. Are you getting the same errors for hw_emu as well?

kaanolgu, 2020-07-20:
I have tried the hw_emu and it gives the same error again. It works fine for sw_emu but hw_emu & hw gives error

virata-xilinx, 2020-07-20:
Cloned the 2019.2 branch and sourced 2019.2. Tried hw_emu for all the DATA_SIZES i.e. 16, 32, 64, 128 and 256. 
Getting TEST PASSED for 16, 32 and 64 . Able to reproduce the issue for 128 and 256 (i.e. getting the kernel compilation failure for both).

heeran-xilinx, 2020-07-31:
Closing this as no response yet to last comment. Please re-open if it is still an issue and provide more information to reproduce the issue.

closed 2020-07-31 by heeran-xilinx