Xilinx/Vitis_Accel_Examples #17: Resource utilization: RAMB18 and RAMB36/FIFO over-utilized in Pblock pblock_dynamic_region

kaanolgu, 2020-07-29:
Hello,
I am trying to compile the array_partition example with size modified to 1024x1024 matrix multiplication with Vitis 2020.1 software for Alveo U200 board. I got the following error during the compilation for hardware: 

```
===>The following messages were generated while processing /data/master/Vitis_Accel_Examples/cpp_kernels/array_partition/build/link/vivado/vpl/prj/prj.runs/impl_1 :
ERROR: [VPL UTLZ-1] Resource utilization: RAMB18 and RAMB36/FIFO over-utilized in Pblock pblock_dynamic_region (This design requires more RAMB18 and RAMB36/FIFO cells than are available in Pblock 'pblock_dynamic_region'. This design requires 10656 of such cell types but only 3696 compatible sites are available in Pblock 'pblock_dynamic_region'. Please consider increasing the span of Pblock 'pblock_dynamic_region' or removing cells from it.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB18 and RAMB36/FIFO over-utilized in Top Level Design (This design requires more RAMB18 and RAMB36/FIFO cells than are available in the target device. This design requires 11296 of such cell types but only 4320 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB36/FIFO over-utilized in Pblock pblock_dynamic_region (This design requires more RAMB36/FIFO cells than are available in Pblock 'pblock_dynamic_region'. This design requires 5328 of such cell types but only 1848 compatible sites are available in Pblock 'pblock_dynamic_region'. Please consider increasing the span of Pblock 'pblock_dynamic_region' or removing cells from it.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB36/FIFO over-utilized in Top Level Design (This design requires more RAMB36/FIFO cells than are available in the target device. This design requires 5639 of such cell types but only 2160 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB36E2 over-utilized in Pblock pblock_dynamic_region (This design requires more RAMB36E2 cells than are available in Pblock 'pblock_dynamic_region'. This design requires 5328 of such cell types but only 1848 compatible sites are available in Pblock 'pblock_dynamic_region'. Please consider increasing the span of Pblock 'pblock_dynamic_region' or removing cells from it.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB36E2 over-utilized in Top Level Design (This design requires more RAMB36E2 cells than are available in the target device. This design requires 5639 of such cell types but only 2160 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
ERROR: [VPL 4-23] Error(s) found during DRC. Placer not run.
ERROR: [VPL 60-773] In '/data/master/Vitis_Accel_Examples/cpp_kernels/array_partition/build/link/vivado/vpl/runme.log', caught Tcl error:  problem implementing dynamic region, impl_1: place_design ERROR, please look at the run log file '/data/master/Vitis_Accel_Examples/cpp_kernels/array_partition/build/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
WARNING: [VPL 60-732] Link warning: The available BRAMs may not be sufficient to accommodate the kernels
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, impl_1: place_design ERROR, please look at the run log file '/data/master/Vitis_Accel_Examples/cpp_kernels/array_partition/build/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [10:55:44] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:45 ; elapsed = 01:09:52 . Memory (MB): peak = 1323.551 ; gain = 0.000 ; free physical = 38381 ; free virtual = 177854
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
Makefile:118: recipe for target 'build//matmul.xclbin' failed
make: *** [build//matmul.xclbin] Error 1
nimbix@JARVICENAE-0A0A1849:~/data/master/Vitis_Accel_Examples/cpp_kernels/array_partition$
```

what could be the possible solution for this issue ? 
Thank you!

virata-xilinx, 2020-07-29:
Hi,

I will try to reproduce the issue. Can you please specify which VITIS and XRT version did you source?

kaanolgu, 2020-07-30:
Hello,

XRT build version: 2.6.655
VITIS 2020.1, array_partition example from the master branch of Vitis_Accel_Examples

heeran-xilinx, 2020-07-31:
Hi @kaanolgu ,
As you have increased the data size to large, it reports following errors in log:
ERROR: [VPL UTLZ-1] Resource utilization: RAMB18 and RAMB36/FIFO over-utilized in Pblock pblock_dynamic_region (This design requires more RAMB18 and RAMB36/FIFO cells than are available in Pblock 'pblock_dynamic_region'. This design requires 10656 of such cell types but only 3696 compatible sites are available in Pblock 'pblock_dynamic_region'. Please consider increasing the span of Pblock 'pblock_dynamic_region' or removing cells from it.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB18 and RAMB36/FIFO over-utilized in Top Level Design (This design requires more RAMB18 and RAMB36/FIFO cells than are available in the target device. This design requires 11296 of such cell types but only 4320 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB36/FIFO over-utilized in Pblock pblock_dynamic_region (This design requires more RAMB36/FIFO cells than are available in Pblock 'pblock_dynamic_region'. This design requires 5328 of such cell types but only 1848 compatible sites are available in Pblock 'pblock_dynamic_region'. Please consider increasing the span of Pblock 'pblock_dynamic_region' or removing cells from it.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB36/FIFO over-utilized in Top Level Design (This design requires more RAMB36/FIFO cells than are available in the target device. This design requires 5639 of such cell types but only 2160 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB36E2 over-utilized in Pblock pblock_dynamic_region (This design requires more RAMB36E2 cells than are available in Pblock 'pblock_dynamic_region'. This design requires 5328 of such cell types but only 1848 compatible sites are available in Pblock 'pblock_dynamic_region'. Please consider increasing the span of Pblock 'pblock_dynamic_region' or removing cells from it.)
ERROR: [VPL UTLZ-1] Resource utilization: RAMB36E2 over-utilized in Top Level Design (This design requires more RAMB36E2 cells than are available in the target device. This design requires 5639 of such cell types but only 2160 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
ERROR: [VPL 4-23] Error(s) found during DRC. Placer not run.
ERROR: [VPL 60-773] In '/data/master/Vitis_Accel_Examples/cpp_kernels/array_partition/build/link/vivado/vpl/runme.log', caught Tcl error:  problem implementing dynamic region, impl_1: place_design ERROR, please look at the run log file '/data/master/Vitis_Accel_Examples/cpp_kernels/array_partition/build/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.

Please do use Xilinx Forum for such queries:
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v

Please use this Git Issues  if you are facing any issue with any existing design without any change in example.

Heera

closed 2020-07-31 by heeran-xilinx

virata-xilinx, 2020-07-31:
I experimented a bit with the matrix sizes and observed that the example was working fine till a matrix size of 256x256. Above that the issue is reproducible.