Xilinx/Vitis_Accel_Examples #18: Segmentation fault during build

zhenhaohe, 2020-07-30:
Hi, I was building the rtl_vadd example code with 2019.2 branch and there is segmentation fault error. 

The detail command and the log output is the following:

make all TARGET=hw DEVICE=/opt/xilinx/platforms/xilinx_u280_xdma_201920_3/xilinx_u280_xdma_201920_3.xpfm 

g++ -I../..//common/includes/xcl2 -I/opt/xilinx/xrt/include -I/opt/Xilinx/Vivado/2019.2/include -Wall -O0 -g -std=c++11 -fmessage-length=0 ../..//common/includes/xcl2/xcl2.cpp src/host.cpp   -o 'host'  -L/opt/xilinx/xrt/lib -lOpenCL -lpthread  -lrt -lstdc++ 
mkdir -p ./_x.hw.xilinx_u280_xdma_201920_3
/opt/Xilinx/Vivado/2019.2/bin/vivado -mode batch -source scripts/gen_xo.tcl -tclargs ./_x.hw.xilinx_u280_xdma_201920_3/vadd.xo vadd hw /opt/xilinx/platforms/xilinx_u280_xdma_201920_3/xilinx_u280_xdma_201920_3.xpfm xilinx_u280_xdma_201920_3

****** Vivado v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
  **** IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source scripts/gen_xo.tcl
# if { $::argc != 5 } {
#     puts "ERROR: Program \"$::argv0\" requires 5 arguments!\n"
#     puts "Usage: $::argv0 <xoname> <krnl_name> <target> <xpfm_path> <device>\n"
#     exit
# }
# set xoname  [lindex $::argv 0]
# set krnl_name [lindex $::argv 1]
# set target    [lindex $::argv 2]
# set xpfm_path [lindex $::argv 3]
# set device    [lindex $::argv 4]
# set suffix "${krnl_name}_${target}_${device}"
# source -notrace ./scripts/package_kernel.tcl
INFO: [IP_Flow 19-5169] Module 'krnl_vadd_rtl' uses SystemVerilog sources with a Verilog top file. These SystemVerilog files will not be analysed by the packager.
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2019.2/data/ip'.
INFO: [IP_Flow 19-5107] Inferred bus interface 'm_axi_gmem' of definition 'xilinx.com:interface:aximm:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-5107] Inferred bus interface 's_axi_control' of definition 'xilinx.com:interface:aximm:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-5107] Inferred bus interface 'ap_rst_n' of definition 'xilinx.com:signal:reset:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-5107] Inferred bus interface 'ap_clk' of definition 'xilinx.com:signal:clock:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-5107] Inferred bus interface 'interrupt' of definition 'xilinx.com:signal:interrupt:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-4728] Bus Interface 'ap_rst_n': Added interface parameter 'POLARITY' with value 'ACTIVE_LOW'.
INFO: [IP_Flow 19-4728] Bus Interface 'interrupt': Added interface parameter 'SENSITIVITY' with value 'LEVEL_HIGH'.
INFO: [IP_Flow 19-4728] Bus Interface 'ap_clk': Added interface parameter 'ASSOCIATED_RESET' with value 'ap_rst_n'.
WARNING: [IP_Flow 19-3158] Bus Interface 'm_axi_gmem': FREQ_HZ bus parameter missing from AXI interface when interface is not associated to a clock.
WARNING: [IP_Flow 19-3158] Bus Interface 's_axi_control': FREQ_HZ bus parameter missing from AXI interface when interface is not associated to a clock.
WARNING: [IP_Flow 19-3157] Bus Interface 'ap_rst_n': Bus parameter POLARITY is ACTIVE_LOW but port 'ap_rst_n' is not *resetn - please double check the POLARITY setting.
WARNING: [IP_Flow 19-3153] Bus Interface 'ap_clk': ASSOCIATED_BUSIF bus parameter is missing.
INFO: [IP_Flow 19-2181] Payment Required is not set for this core.
INFO: [IP_Flow 19-2187] The Product Guide file is missing.
INFO: [IP_Flow 19-795] Syncing license key meta-data
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/zhe/Vitis_Accel_Examples/rtl_kernels/rtl_vadd/packaged_kernel_vadd_hw_xilinx_u280_xdma_201920_3/src/krnl_vadd_rtl_control_s_axi.v:]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/zhe/Vitis_Accel_Examples/rtl_kernels/rtl_vadd/packaged_kernel_vadd_hw_xilinx_u280_xdma_201920_3/src/krnl_vadd_rtl_adder.sv:]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/zhe/Vitis_Accel_Examples/rtl_kernels/rtl_vadd/packaged_kernel_vadd_hw_xilinx_u280_xdma_201920_3/src/krnl_vadd_rtl_axi_read_master.sv:]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/zhe/Vitis_Accel_Examples/rtl_kernels/rtl_vadd/packaged_kernel_vadd_hw_xilinx_u280_xdma_201920_3/src/krnl_vadd_rtl_axi_write_master.sv:]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/zhe/Vitis_Accel_Examples/rtl_kernels/rtl_vadd/packaged_kernel_vadd_hw_xilinx_u280_xdma_201920_3/src/krnl_vadd_rtl_counter.sv:]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/zhe/Vitis_Accel_Examples/rtl_kernels/rtl_vadd/packaged_kernel_vadd_hw_xilinx_u280_xdma_201920_3/src/krnl_vadd_rtl_int.sv:]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/zhe/Vitis_Accel_Examples/rtl_kernels/rtl_vadd/packaged_kernel_vadd_hw_xilinx_u280_xdma_201920_3/src/krnl_vadd_rtl.v:]
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2019.2/data/ip'.
INFO: [IP_Flow 19-4728] Bus Interface 'ap_clk': Added interface parameter 'ASSOCIATED_BUSIF' with value 'm_axi_gmem'.
WARNING: [IP_Flow 19-3157] Bus Interface 'ap_rst_n': Bus parameter POLARITY is ACTIVE_LOW but port 'ap_rst_n' is not *resetn - please double check the POLARITY setting.
INFO: [IP_Flow 19-2181] Payment Required is not set for this core.
INFO: [IP_Flow 19-2187] The Product Guide file is missing.
INFO: [Ipptcl 7-1486] check_integrity: Integrity check passed.
# if {[file exists "${xoname}"]} {
#     file delete -force "${xoname}"
# }
# package_xo -xo_path ${xoname} -kernel_name krnl_vadd_rtl -ip_directory ./packaged_kernel_${suffix} -kernel_xml ./src/kernel.xml
WARNING: [Vivado 12-4404] The CPU emulation flow in v++ is only supported when using a packaged XO file that contains C-model files, none were found.
WARNING: [Vivado 12-7038] The bus parameter 'ap_clk.FREQ_HZ' has been detected in the kernel IP 'krnl_vadd_rtl'. This may cause validation errors in v++ if a target platform has a different default kernel clock frequency. If validation errors do occur in v++, please edit and re-generate the kernel IP 'krnl_vadd_rtl' to remove the bus parameter 'ap_clk.FREQ_HZ'
INFO: [Common 17-206] Exiting Vivado at Thu Jul 30 17:41:30 2020...
mkdir -p ./build_dir.hw.xilinx_u280_xdma_201920_3
[... 101 lines cut ...]
[17:58:56] Phase 1 Retarget
[17:59:28] Phase 2 Constant propagation
[17:59:28] Phase 3 Sweep
[18:00:01] Phase 4 BUFG optimization
[18:00:01] Phase 5 Shift Register Optimization
[18:00:01] Phase 6 Post Processing Netlist
[18:00:33] Finished 3rd of 6 tasks (FPGA logic optimization). Elapsed time: 00h 02m 10s 

[18:00:33] Starting logic placement..
[18:01:06] Phase 1 Placer Initialization
[18:01:06] Phase 1.1 Placer Initialization Netlist Sorting
[18:03:16] Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device
[18:04:21] Phase 1.3 Build Placer Netlist Model
[18:05:26] Phase 1.4 Constrain Clocks/Macros
[18:05:26] Phase 2 Global Placement
[18:05:26] Phase 2.1 Floorplanning
[18:06:31] Phase 2.2 Global Placement Core
/opt/Xilinx/Vitis/2019.2/bin/loader: line 280: 72377 Segmentation fault      "$RDI_PROG" "$@"
INFO: [v++ 60-1442] [18:07:21] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:22 ; elapsed = 00:24:55 . Memory (MB): peak = 679.508 ; gain = 0.000 ; free physical = 336846 ; free virtual = 457622
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
Makefile:98: recipe for target 'build_dir.hw.xilinx_u280_xdma_201920_3/vadd.xclbin' failed
make: *** [build_dir.hw.xilinx_u280_xdma_201920_3/vadd.xclbin] Error 1

virata-xilinx, 2020-08-04:
Hi,

Can you please mention which VITIS and XRT did you source. I will try to reproduce the issue.

zhenhaohe, 2020-08-04:
Hi the Vitis version is 2019.2 the platform is xilinx_u280_xdma_201920_3

virata-xilinx, 2020-08-05:
Hi,

Sourced 2019.2 Vitis and XRT and ran for xilinx_u280_xdma_201920_3 and the issue is not reproducible. Can you please do a make cleanall try to re-build the example.

heeran-xilinx, 2020-08-17:
closing this issue as no  respond for last 10 days. Please re-open if issue still persist and provide more details to reproduce.

closed 2020-08-17 by heeran-xilinx