Xilinx/Vitis_Accel_Examples #19: Build Vitis_Accel_Examples/sys_opt/multiple_process/ project fail

lmaxeniro, 2020-08-13:
For this project: Vitis_Accel_Examples/sys_opt/multiple_process/
When bulid the project from command line: the three xo files can be generated, but during the link stage, error report:
---------------
INFO: [SYSTEM_LINK 82-70] Extracting xo v3 file /home/vitis_workspace_new/Vitis_Accel_Examples/sys_opt/multiple_process/_x.hw.xilinx_u200_xdma_201830_2/krnl_vadd.xo
ERROR: [KernelCheck 83-117] '' kernel.xml indicates the existence of port , which does not exist
ERROR: [SYSTEM_LINK 82-63] Error generating intermediate file /home/vitis_workspace_new/Vitis_Accel_Examples/sys_opt/multiple_process/build_dir.hw.xilinx_u200_xdma_201830_2/link/sys_link/iprepo/temp/xo0/ip_repo/xilinx_com_hls_krnl_vadd_1_0/krnl_vadd.fcnmap.xml
ERROR: [SYSTEM_LINK 82-78] Unable to create function map for HLS kernel krnl_vadd
ERROR: [SYSTEM_LINK 82-84] Unable to process .xo file: /home/vitis_workspace_new/Vitis_Accel_Examples/sys_opt/multiple_process/_x.hw.xilinx_u200_xdma_201830_2/krnl_vadd.xo
ERROR: [SYSTEM_LINK 82-66] Error processing .xo files
ERROR: [SYSTEM_LINK 82-100] Error processing object files, exiting
INFO: [v++ 60-1442] [09:46:33] Run run_link: Step system_link: Failed
Time (s): cpu = 00:00:00.84 ; elapsed = 00:00:00.92 . Memory (MB): peak = 674.820 ; gain = 0.000 ; free physical = 50142 ; free virtual = 61457
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking

virata-xilinx, 2020-08-13:
Hi,

Can you please mention which VITIS and XRT did you source. I will try to reproduce the issue.

virata-xilinx, 2020-08-13:
Hi,

I sourced 2020.1 VITIS and XRT and did a hw build for u200_xdma_201830_@ and the issue is not reproducible. Can you please re-buildteh example once after doing a make cleanall.

lmaxeniro, 2020-08-14:
If I use Vitis IDE, the build is OK. 
But under the command line the above will be fail, maybe it is due to some ENV missing? I have some other rtl project build under command line without problem then..
Here is my VITIS version:
`
****** Vitis v2019.2.1 (64-bit)
**** SW Build 2729669 on Thu Dec  5 04:48:51 MST 2019
`
XRT version
`  "BUILD_VERSION" : "2.6.655",
  "BUILD_VERSION_DATE" : "Fri, 22 May 2020 12:05:03 -0700",
  "BUILD_BRANCH" : "2020.1",
  "VERSION_HASH" : "2d6bfe4ce91051d4e5b499d38fc493586dd4859a",
  "VERSION_HASH_DATE" : "Fri, 22 May 2020 20:53:33 +0530"
`

virata-xilinx, 2020-08-14:
I also tested on command line only.
I see your VITIS version is 2019.2 but your XRT version is 2020.1. Which branch are you running for? Is it 2019.2 or master?

lmaxeniro, 2020-08-17:
I think I am using master..
I think I can close this tag--anyway it works on Vitis IDE.

closed 2020-08-17 by lmaxeniro