Xilinx/Vitis_Accel_Examples #20: Vitis_Accel_Examples/sys_opt/advanced_config has two clock setup?

lmaxeniro, 2020-08-13:
1. in the 1st config file (used for kernel compile): https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/sys_opt/advanced_config/vadd_vadd.ini
kernel_frequency=0:150
[advanced]
prop=solution.hls_pre_tcl=my_directives.tcl

2. in 2nd config file(used for linker): https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/sys_opt/advanced_config/vadd.ini
kernel_frequency=0:280

Is this correct config?

virata-xilinx, 2020-08-17:
During the kernel compilation the clock frequency will be set at 150 MHz but when the example moves to the system linking stage, the clock frequency will be set at 280 MHz. So, both are different stages with different functionalities and in this case are working at different clock frequencies as well.

lmaxeniro, 2020-08-19:
@virata-xilinx On fpga card if there is only one clock (clock index 0) then what will happen? from my understanding, you can not specify two clocks.. so what will be the finalized kernel clock?

heeran-xilinx, 2020-08-19:
@lmaxeniro,
kernel clock during compilation is only for HLS code compilation. Based on kernel code frequency, HLS try to generate RTL.  It can re-structure the generated state machine and data-path to meet the timing. This is not the actual system clock.
Kernel clock during linker is actually a system clock, which will be used by Vivado Synthesis, Placement & Route tool to meet the timing.  Linker time clock frequency will be the actual system clock on which fpga card will be running.

lmaxeniro, 2020-08-19:
@heeran-xilinx Thanks for the explanation. If the final kernel clock is decided in link stage (280M), then what is the benefit to have an (even lower 150M) target in compiling stage?

heeran-xilinx, 2020-08-19:
Hi @lmaxeniro ,
Yes I agree. We are going to update example and keeping both same.

lmaxeniro, 2020-08-19:
So for my test--can I just change both to the same value, say 280MHz?

vishnuchebrolu, 2020-08-19:
Yes, you can go ahead and update with 280 in both the places.

closed 2020-08-19 by lmaxeniro

lmaxeniro, 2020-08-19:
BTW, can you please help to explain the meaning of the timing constrain inside [my_directives]:([https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/sys_opt/advanced_config/my_directives.tcl])
set_clock_uncertainty 0.370 how that value is calculated (I think it shall be relevant to the target clock 280M, for exp?)
And furthermore -- I have one rtl design, can I use the same optimization as this (This design is using HLS to compile a cpp design).

heeran-xilinx, 2020-08-19:
set_clock_uncertainty is one of configuration parameter for HLS kernel. Using this example, we are explaining how an  user can customize this clock uncertainty. 
clock uncertainty is not applicable for RTL kernel. For RTL case, only linker clock setting is meaningful to set target card frequency.