Xilinx/Vitis_Accel_Examples #21: Perl not found with "hello_world" example with ZCU102 device

linusyukwong, 2020-08-18:
Has anyone gone through the "hello_world" example with ZCU102 device? I am running ubuntu 16.04.6 with Vitis, Vivado, PetaLinux 2020.1 all installed under /tools/Xilinx/

After make "all" with: make all TARGET=sw_emu DEVICE=xilinx_zcu102_base_202010_1 HOST_ARCH=aarch64 SYSROOT=~/xilinx-zynqmp-common-v2020.1/ir/sysroots/aarch64-xilinx-linux

I try to run make "check: with: make check TARGET=sw_emu DEVICE=xilinx_zcu102_base_202010_1 HOST_ARCH=aarch64 SYSROOT=~/xilinx-zynqmp-common-v2020.1/ir/sysroots/aarch64-xilinx-linux

but it outputs: 
/bin/sh: /home/ywongai/Vitis_Accel_Examples/common/utility/run_emulation.pl: /tools/xgs/perl/5.8.5/bin/perl: bad interpreter: No such file or directory

Full output of make check:
ywongai@linuxhost:~/Vitis_Accel_Examples/hello_world$ make check TARGET=sw_emu DEVICE=xilinx_zcu102_base_202010_1 HOST_ARCH=aarch64 SYSROOT=/home/ywongai/xilinx-zynqmp-common-v2020.1/ir/sysroots/aarch64-xilinx-linux
/tools/Xilinx/Vitis/2020.1/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-g++ -I..//common/includes/xcl2 -I/home/ywongai/xilinx-zynqmp-common-v2020.1/ir/sysroots/aarch64-xilinx-linux/usr/include/xrt -I/tools/Xilinx/Vivado/2020.1/include -Wall -O0 -g -std=c++11 -fmessage-length=0 ..//common/includes/xcl2/xcl2.cpp src/host.cpp  -o 'host'  -L/home/ywongai/xilinx-zynqmp-common-v2020.1/ir/sysroots/aarch64-xilinx-linux/usr/lib -lOpenCL -lpthread  -lrt -lstdc++  --sysroot=/home/ywongai/xilinx-zynqmp-common-v2020.1/ir/sysroots/aarch64-xilinx-linux
rm -rf run_app.sh
v++ -t sw_emu --platform xilinx_zcu102_base_202010_1 -p ./build_dir.sw_emu.xilinx_zcu102_base_202010_1/vadd.xclbin --package.out_dir package.sw_emu --package.rootfs /home/ywongai/xilinx-zynqmp-common-v2020.1/rootfs.ext4 --package.sd_file /home/ywongai/xilinx-zynqmp-common-v2020.1/Image --package.sd_file xrt.ini --package.sd_file run_app.sh --package.sd_file host -o vadd.xclbin
Option Map File Used: '/tools/Xilinx/Vitis/2020.1/data/vitis/vpp/optMap.xml'

****** v++ v2020.1 (64-bit)
  **** SW Build 2902540 on Wed May 27 19:54:35 MDT 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ package can be found at:
        Reports: /home/ywongai/Vitis_Accel_Examples/hello_world/_x/reports/package
        Log files: /home/ywongai/Vitis_Accel_Examples/hello_world/_x/logs/package
INFO: [v++ 60-1657] Initializing dispatch client.
Running Dispatch Server on port:34295
INFO: [v++ 60-1548] Creating build summary session with primary output /home/ywongai/Vitis_Accel_Examples/hello_world/vadd.xclbin.package_summary, at Tue Aug 18 12:34:56 2020
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Tue Aug 18 12:34:56 2020
Running Rule Check Server on port:34423
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/ywongai/Vitis_Accel_Examples/hello_world/_x/reports/package/v++_package_vadd_guidance.html', at Tue Aug 18 12:34:57 2020
INFO: [v++ 60-895]   Target platform: /tools/Xilinx/Vitis/2020.1/platforms/xilinx_zcu102_base_202010_1/xilinx_zcu102_base_202010_1.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/tools/Xilinx/Vitis/2020.1/platforms/xilinx_zcu102_base_202010_1/hw/xilinx_zcu102_base_202010_1.xsa'
INFO: [v++ 60-2256] Packaging for software emulation
WARNING: [v++ 82-1077] Kernel image is not specified for linux domain
WARNING: [v++ 82-1097] Skipping boot image generation, since FSBL is directly loaded onto emulator
INFO: [v++ 82-1017] Deleting existing qemu_args.txt
INFO: [v++ 82-1016] Deleting existing pmc_args.txt
INFO: [v++ 82-1014] Deleting existing launch_hw_emulator.sh
INFO: [v++ 82-1022] Generating /home/ywongai/Vitis_Accel_Examples/hello_world/package.sw_emu/launch_sw_emu.sh
INFO: [v++ 82-1018] Deleting existing sd_card directory
INFO: [v++ 82-1011] Creating sd_card directory
INFO: [v++ 82-1075] Deleting the existing directory /home/ywongai/Vitis_Accel_Examples/hello_world/package.sw_emu/sd_card//data/emulation/unified
INFO: [v++ 82-1083] Deleting existing sd_card.img
FATSIZE:1124
fat_start:63
fat_end:2096639
fat_sector:2096577
ext4_start:0
ext4_sector:4142856
EXT4SIZE:2048
TOTALSIZE:3172
dummy_ext4_sector:51448
sd_card_fat_start:2048
sd_card_ext4_start:2195456
dummy_ext4_start:6338312
2096577+0 records in
2096577+0 records out
1073447424 bytes (1.1 GB, 1.0 GiB) copied, 6.1871 s, 173 MB/s
4142856+0 records in
4142856+0 records out
2121142272 bytes (2.1 GB, 2.0 GiB) copied, 11.088 s, 191 MB/s
51448+0 records in
51448+0 records out
26341376 bytes (26 MB, 25 MiB) copied, 0.129789 s, 203 MB/s
INFO: [v++ 60-2343] Use the vitis_analyzer tool to visualize and navigate the relevant reports. Run the following command.
    vitis_analyzer /home/ywongai/Vitis_Accel_Examples/hello_world/vadd.xclbin.package_summary
INFO: [v++ 60-791] Total elapsed time: 0h 0m 33s
INFO: [v++ 60-1653] Closing dispatch client.
/home/ywongai/Vitis_Accel_Examples/common/utility/run_emulation.pl "./package.sw_emu/launch_sw_emu.sh | tee run_app.log" "./run_app.sh sw_emu" "TEST PASSED" "7"
/bin/sh: /home/ywongai/Vitis_Accel_Examples/common/utility/run_emulation.pl: /tools/xgs/perl/5.8.5/bin/perl: bad interpreter: No such file or directory
make: *** [Makefile:132: check] Error 126

virata-xilinx, 2020-08-18:
Hi,

Can you please ensure two things -
1. Instead of SYSROOT mention the path to EDGE_COMMON_SW. The SYSROOT and SD_IMAGE_FILE are derived from that. You can use the "make help" command to see the same.
2. Kindly ensure that perl is properly installed on the machine.

linusyukwong, 2020-08-18:
Thank you very much for your reply.

1. I have exported the EDGE_COMMON_SW. So there is no need to specify SYSROOT anymore in 2020.1?

2. perl is properly installed at /usr/bin/perl. However, when I examine common/utility/run_emulation.pl, it seems to assume that perl is installed at /tools/xgs/perl/5.8.5/bin/perl? I am not familiar with perl so I may be wrong, but there is no /tools/xgs directory in my machine.

virata-xilinx, 2020-08-18:
Hi,

1. Yes, since both the SYSROOT and SD_IMAGE_FILE paths are determined in the utils.mk using the EDGE_COMMON_SW variable, you only need to specify EDGE_COMMON_SW.
2. I guess we have put our internal perl installation path in the script . Can you please try replacing the first line of common/utility/run_emulation.pl with your installed path.

heeran-xilinx, 2020-08-18:
Could you please try again after replacing  "#!/tools/xgs/perl/5.8.5/bin/perl" to "#!/usr/bin/perl" here?
https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/common/utility/run_emulation.pl#L1

linusyukwong, 2020-08-18:
Here is the output after replacing "#!/tools/xgs/perl/5.8.5/bin/perl" to "#!/usr/bin/perl". It seems that the qemu has exited abnormally. I am still specifying SYSROOT because it doesn't run if I didn't specify.

ywongai@linuxhost:~/Vitis_Accel_Examples/hello_world$ make check TARGET=sw_emu DEVICE=xilinx_zcu102_base_202010_1 HOST_ARCH=aarch64 SYSROOT=/home/ywongai/xilinx-zynqmp-common-v2020.1/ir/sysroots/aarch64-xilinx-linux
/tools/Xilinx/Vitis/2020.1/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-g++ -I..//common/includes/xcl2 -I/home/ywongai/xilinx-zynqmp-common-v2020.1/ir/sysroots/aarch64-xilinx-linux/usr/include/xrt -I/tools/Xilinx/Vivado/2020.1/include -Wall -O0 -g -std=c++11 -fmessage-length=0 ..//common/includes/xcl2/xcl2.cpp src/host.cpp  -o 'host'  -L/home/ywongai/xilinx-zynqmp-common-v2020.1/ir/sysroots/aarch64-xilinx-linux/usr/lib -lOpenCL -lpthread  -lrt -lstdc++  --sysroot=/home/ywongai/xilinx-zynqmp-common-v2020.1/ir/sysroots/aarch64-xilinx-linux
rm -rf run_app.sh
v++ -t sw_emu --platform xilinx_zcu102_base_202010_1 -p ./build_dir.sw_emu.xilinx_zcu102_base_202010_1/vadd.xclbin --package.out_dir package.sw_emu --package.rootfs /home/ywongai/xilinx-zynqmp-common-v2020.1/rootfs.ext4 --package.sd_file /home/ywongai/xilinx-zynqmp-common-v2020.1/Image --package.sd_file xrt.ini --package.sd_file run_app.sh --package.sd_file host -o vadd.xclbin
Option Map File Used: '/tools/Xilinx/Vitis/2020.1/data/vitis/vpp/optMap.xml'

****** v++ v2020.1 (64-bit)
  **** SW Build 2902540 on Wed May 27 19:54:35 MDT 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ package can be found at:
        Reports: /home/ywongai/Vitis_Accel_Examples/hello_world/_x/reports/package
        Log files: /home/ywongai/Vitis_Accel_Examples/hello_world/_x/logs/package
INFO: [v++ 60-1657] Initializing dispatch client.
Running Dispatch Server on port:46353
INFO: [v++ 60-1548] Creating build summary session with primary output /home/ywongai/Vitis_Accel_Examples/hello_world/vadd.xclbin.package_summary, at Tue Aug 18 21:29:22 2020
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Tue Aug 18 21:29:22 2020
Running Rule Check Server on port:39705
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/ywongai/Vitis_Accel_Examples/hello_world/_x/reports/package/v++_package_vadd_guidance.html', at Tue Aug 18 21:29:24 2020
INFO: [v++ 60-895]   Target platform: /tools/Xilinx/Vitis/2020.1/platforms/xilinx_zcu102_base_202010_1/xilinx_zcu102_base_202010_1.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/tools/Xilinx/Vitis/2020.1/platforms/xilinx_zcu102_base_202010_1/hw/xilinx_zcu102_base_202010_1.xsa'
INFO: [v++ 60-2256] Packaging for software emulation
WARNING: [v++ 82-1077] Kernel image is not specified for linux domain
WARNING: [v++ 82-1097] Skipping boot image generation, since FSBL is directly loaded onto emulator
INFO: [v++ 82-1017] Deleting existing qemu_args.txt
INFO: [v++ 82-1016] Deleting existing pmc_args.txt
INFO: [v++ 82-1014] Deleting existing launch_hw_emulator.sh
INFO: [v++ 82-1022] Generating /home/ywongai/Vitis_Accel_Examples/hello_world/package.sw_emu/launch_sw_emu.sh
INFO: [v++ 82-1018] Deleting existing sd_card directory
INFO: [v++ 82-1011] Creating sd_card directory
INFO: [v++ 82-1075] Deleting the existing directory /home/ywongai/Vitis_Accel_Examples/hello_world/package.sw_emu/sd_card//data/emulation/unified
INFO: [v++ 82-1083] Deleting existing sd_card.img
FATSIZE:1124
fat_start:63
fat_end:2096639
fat_sector:2096577
ext4_start:0
ext4_sector:4142856
EXT4SIZE:2048
TOTALSIZE:3172
dummy_ext4_sector:51448
sd_card_fat_start:2048
sd_card_ext4_start:2195456
dummy_ext4_start:6338312
2096577+0 records in
2096577+0 records out
1073447424 bytes (1.1 GB, 1.0 GiB) copied, 5.37645 s, 200 MB/s
4142856+0 records in
4142856+0 records out
2121142272 bytes (2.1 GB, 2.0 GiB) copied, 10.2972 s, 206 MB/s
51448+0 records in
51448+0 records out
26341376 bytes (26 MB, 25 MiB) copied, 0.162979 s, 162 MB/s
INFO: [v++ 60-2343] Use the vitis_analyzer tool to visualize and navigate the relevant reports. Run the following command.
    vitis_analyzer /home/ywongai/Vitis_Accel_Examples/hello_world/vadd.xclbin.package_summary
INFO: [v++ 60-791] Total elapsed time: 0h 0m 31s
INFO: [v++ 60-1653] Closing dispatch client.
/home/ywongai/Vitis_Accel_Examples/common/utility/run_emulation.pl "./package.sw_emu/launch_sw_emu.sh | tee run_app.log" "./run_app.sh sw_emu" "TEST PASSED" "7"
Current working dir /home/ywongai/Vitis_Accel_Examples/hello_world/package.sw_emu
Required emulation files like qemu_args exists
Running SW Emulation
INFO :  [LAUNCH_EMULATOR] pl_sim_dir option is not provided.
Starting QEMU
 - Press <Ctrl-a h> for help
Waiting for QEMU to start. qemu_port 9277
Qemu_pids 9077 9078
QEMU started. qemu_pid=9077.
Waiting for PMU to start. Qemu_pids 9131 9132
PMC started. pmc_pid=9131
Starting PL simulation.Generating PLLauncher commandline
running PLL Launcher
[LAUNCH_EMULATOR] INFO: 21:29:50 : PS-QEMU exited
[LAUNCH_EMULATOR] INFO: 21:29:52 : PS-QEMU exited
[LAUNCH_EMULATOR] INFO: 21:29:52 : PMU/PMC-QEMU exited
[LAUNCH_EMULATOR] INFO: 21:29:52 : Trying to kill PLLauncher forcefully because QEMU exited
[LAUNCH_EMULATOR] INFO: 21:29:52 : PLLauncher exited by force
Please refer PS /simulate logs at /home/ywongai/Vitis_Accel_Examples/hello_world/package.sw_emu for more details.
DONE!
INFO: Emulation ran successfully

heeran-xilinx, 2020-08-18:
Thanks for trying this.  Seems like perl script is still not working correctly in your setup.
Could you please let us know what is perl version? which OS you are running this application?

linusyukwong, 2020-08-18:
I am running ubuntu 16.04.6
The perl version is v5.22.1

heeran-xilinx, 2020-08-19:
Thanks for sharing more details.
If possible, could you please try below steps?

1.	./package.sw_emu/launch_sw_emu.sh
2.	Once linux boots, then on embedded linux prompt inside qemu, run “./run_app.sh sw_emu” and see if the application prints TEST PASSED.

linusyukwong, 2020-08-19:
On running the launch_sw_emu.sh, I get the following output:

ywongai@linuxhost:~/Vitis_Accel_Examples/hello_world/package.sw_emu$ ./launch_sw_emu.sh
Required emulation files like qemu_args exists
Current working dir /home/ywongai/Vitis_Accel_Examples/hello_world/package.sw_emu
Running SW Emulation
INFO :  [LAUNCH_EMULATOR] pl_sim_dir option is not provided.
Starting QEMU
 - Press <Ctrl-a h> for help
Waiting for QEMU to start. qemu_port 9267
Qemu_pids 24334 24335
QEMU started. qemu_pid=24334.
Waiting for PMU to start. Qemu_pids 24388 24389
PMC started. pmc_pid=24388
Starting PL simulation.Generating PLLauncher commandline
running PLL Launcher
[LAUNCH_EMULATOR] INFO: 11:40:35 : PS-QEMU exited
[LAUNCH_EMULATOR] INFO: 11:40:37 : PS-QEMU exited
[LAUNCH_EMULATOR] INFO: 11:40:37 : PMU/PMC-QEMU exited
[LAUNCH_EMULATOR] INFO: 11:40:37 : Trying to kill PLLauncher forcefully because QEMU exited
[LAUNCH_EMULATOR] INFO: 11:40:37 : PLLauncher exited by force
Please refer PS /simulate logs at /home/ywongai/Vitis_Accel_Examples/hello_world/package.sw_emu for more details.
DONE!
INFO: Emulation ran successfully

heeran-xilinx, 2020-08-19:
Thanks @linusyukwong for trying out these steps.  We are discussing this case with Tool team and will give you updates soon.

linusyukwong, 2020-08-19:
Once I change the device to a data center card, I am able to run the sw_emu of the example successfuly. Maybe it is a problem of qemu of PetaLinux.

closed 2020-11-27 by heeran-xilinx