Xilinx/Vitis_Accel_Examples #25: v++ not able to synthesize the memory interface of top-level function

Hecmay, 2020-09-11:
Hi,

I was trying to run the PLRAM example in the repo, and ran into some issues when compiling it with v++. When I compile the source directly, as https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/cpp_kernels/plram_access/src/mmult.cpp#L45-L51, the tool first complains that the size of the input array is unknown, so I changed the top-level function signature as followed:

```c++
void mmult(const int a[64], // Read-Only Matrix A
           const int b[64], // Read-Only Matrix B
           int c[64],       // Output Result
           int a_row,    // Matrix A Row Size
           int a_col,    // Matrix A Col Size
           int b_col     // Matrix B Col Size
           ) {
```
After I changed the function, the v++ complains another issue: 

```shell
===>The following messages were generated while  performing high-level synthesis for kernel: mmult Log file: /work/shared/users/phd/sx233/Vitis_Accel_Examples/cpp_kernels/plram_access/_x.hw.xilinx_u280_xdma_201920_1/mmult/mmult/vivado_hls.log :
INFO: [v++ 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [v++ 204-61] Pipelining loop 'readA'.
INFO: [v++ 204-61] Pipelining result : Target II = 1, Final II = 1, Depth = 2.
INFO: [v++ 204-61] Pipelining loop 'readB'.
INFO: [v++ 204-61] Pipelining result : Target II = 1, Final II = 1, Depth = 2.
INFO: [v++ 204-61] Pipelining loop 'outer_loop1_outer_loop2_inner_loop'.
INFO: [v++ 204-61] Pipelining result : Target II = 1, Final II = 1, Depth = 8.
INFO: [v++ 204-61] Pipelining loop 'writeC'.
INFO: [v++ 204-61] Pipelining result : Target II = 1, Final II = 1, Depth = 2.
INFO: [v++ 200-790] **** Loop Constraint Status: All loop constraints were satisfied.
INFO: [v++ 200-789] **** Estimated Fmax: 309.02 MHz
ERROR: [v++ 213-400] This design has interfaces which are not supported by Vitis. Check interface pragmas to ensure the design only uses one s_axilite interface including the 'return' port and one or more m_axi interfaces with offset=slave set to the s_axilite interface; there should not be any other port interface used.
ERROR: [v++ 200-445] Unexpected error generating RTL model: generate_json error: Found incompatible interfaces
ERROR: [v++ 60-300] Failed to build kernel(ip) mmult, see log for details: /work/shared/users/Vitis_Accel_Examples/cpp_kernels/plram_access/_x.hw.xilinx_u280_xdma_201920_1/mmult/mmult/vivado_hls.log
ERROR: [v++ 60-599] Kernel compilation failed to complete
ERROR: [v++ 60-592] Failed to finish compilation
make: *** [_x.hw.xilinx_u280_xdma_201920_1/mmult.xo] Error 1
```

virata-xilinx, 2020-09-12:
Hi,

Can you please specify which branch are you running it for. Also which VITIS and XRT version did you source. Was the example running for sw_emu? I tried the example for sw_emu and observed that the same is working for 2020.1 for xilinx_u280_xdma_201920_1.

Hecmay, 2020-09-12:
The XRT version:
```shell
{
  "BUILD_VERSION" : "2.3.1301",
  "BUILD_VERSION_DATE" : "Fri, 25 Oct 2019 03:04:42 +0000",
  "BUILD_BRANCH" : "2019.2",
  "VERSION_HASH" : "192e706aea53163a04c574f9b3fe9ed76b6ca471",
  "VERSION_HASH_DATE" : "Thu, 24 Oct 2019 19:27:30 -0700"
}
```

v++ version
```shell
****** v++ v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.
```

 I was compiling for the HW.

virata-xilinx, 2020-09-13:
I tried the example for the master branch sourcing 2020.1 Vitis and XRT. The examples is working for me for all the flows i.e. sw_emu, hw_emu and hw. I ran the example with no changes. 
I have also tried for the 2019.2 branch sourcing 2019.2 Vitis and XRT and both the emulation flows are working for me.
Kindly source the respective Vitis and XRT. Also can you please try the original example (I checked for the original example) first for sw_emu and hw_emu and notify if you are facing any issues.

Hecmay, 2020-09-13:
Thanks, @virata-xilinx! 

I tried the 2019.2 branch and it worked well. So it seems that Vitis 2019.2 requires users to specify the interface programs, otherwise, the synthesis would fail?  

Also, I ran to another (performance) issue with the interface pragma -- when I synthesize an application without the interface pragma using VIvado HLS, the program can achieve II=1. The application is a simple 2D CONV operator. However, when I add the interface pragmas and compile it with v++, the II becomes much worse. And from the HLS log, the tool throws the warning message as follows:

```shell
INFO: [SCHED 204-61] Pipelining loop 'Loop 1.1'.
WARNING: [SCHED 204-69] Unable to schedule bus request on port 'gmem1' (/work/shared/users/phd/sx233/alveo-test/kmeans/kernel.cpp:35) due to limited memory ports. Please consider using a memory core with more ports or partitioning the array.
INFO: [SCHED 204-61] Unable to satisfy pipeline directive: Unable to pipeline the region.
```

heeran-xilinx, 2020-09-14:
HI @Hecmay ,

Thanks for reporting this issue. 
If possible, could you please post your Vitis (or HLS) related queries to Xilinx Public Forum to get quick response from Xilinx Experts and support staff?
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v

-Heera

Hecmay, 2020-09-14:
Thanks @heeran-xilinx. I will try. The technical support on Xilinx forum is far from responsive, and oftentimes my question won't get any response for quite a long time (like around half a year)... that's why I opened issues here.

heeran-xilinx, 2020-09-15:
Hi @Hecmay ,
Thanks for your valuable feedback on Technical support. I will convey this to Xilinx internal. 
If possible, please post this query there and share the Query link here.

-Heera

closed 2020-10-06 by heeran-xilinx