Xilinx/Vitis_Accel_Examples #26: Error in creating xcbin

srinivasans74, 2020-09-17:
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/Failed-to-create-xcbin-for-gemm-operation/td-p/1151871

virata-xilinx, 2020-09-17:
Hi,

Were you able to generated the xclbins for sw_emu and hw_emu. Also can you please mention the command you ran.

srinivasans74, 2020-09-18:
> Hi,
> 
> Were you able to generated the xclbins for sw_emu and hw_emu. Also can you please mention the command you ran.

srinivasans74, 2020-09-18:
> Hi,
> 
> Were you able to generated the xclbins for sw_emu and hw_emu. Also can you please mention the command you ran.

Yes I was able to generate the xclbins for sw_emu and hw_emu . 
I got a .exe file generated along with the xclbins

when I try to use ./exe <path to xclbin> i am getting these errors

(base) [srinivasan@chewbacca hello_world]$ ./hello_world build_dir.sw_emu.xilinx_u200_xdma_201830_2/vadd.xclbin
Found Platform
Platform Name: Xilinx
INFO: Reading build_dir.sw_emu.xilinx_u200_xdma_201830_2/vadd.xclbin
Loading: 'build_dir.sw_emu.xilinx_u200_xdma_201830_2/vadd.xclbin'
Trying to program device[0]: xilinx_u200_xdma_201830_2
XRT build version: 2.6.655
Build hash: 2d6bfe4ce91051d4e5b499d38fc493586dd4859a
Build date: 2020-05-22 19:05:52
Git branch: 2020.1
PID: 1940
UID: 1005
[Fri Sep 18 05:41:40 2020]
HOST: chewbacca.co.it.pt
EXE: /home/srinivasan/workspace/Vitis_Accel_Examples/hello_world/hello_world
[XRT] ERROR: See dmesg log for details. err=-22
[XRT] ERROR: Failed to load xclbin.
Failed to program device[0] with xclbin file!
Failed to program any device found, exit!

heeran-xilinx, 2020-09-21:
Closing this  as the issue is resolved in Forum  discussion.

heeran-xilinx, 2020-09-21:
Hi @srinivasans74 , Please have further discussion  in Xilinx forum if you  still have issue.

-Heera

closed 2020-09-21 by heeran-xilinx