Xilinx/Vitis_Accel_Examples #27: question about concurrent_kernel_execution

good-for-nothing-man, 2020-10-21:
i use the demo of "/host/concurrent_kernel_execution" on the xilinx board -- Alveo U50.
when i make the software emulation , vitis shows the following error.

Found Platform
Platform Name: Xilinx
INFO: Reading ../matrix_ops.xclbin
Loading: '../matrix_ops.xclbin'
Trying to program device[0]: xilinx_u50_xdma_201920_1
ERROR: S_AXI_CONTROL remap entry is absent in xmlbin
ERROR: S_AXI_CONTROL remap entry is absent in xmlbin
ERROR: S_AXI_CONTROL remap entry is absent in xmlbin
XRT build version: 2.3.1301
Build hash: 192e706aea53163a04c574f9b3fe9ed76b6ca471
Build date: 2019-10-25 03:04:42
Git branch: 2019.2
PID: 55240
UID: 1007
[Tue Oct 20 14:44:46 2020]
HOST: B318
EXE: /home/ding/ghh/kernels_demo/demo/Emulation-SW/demo
[XRT] ERROR: Failed to load xclbin.
Failed to program device[0] with xclbin file!
Failed to program any device found, exit!

i don't understand where the errors they are , so i want to know how to deal with it.

thanks a lot!

vishnuchebrolu, 2020-10-21:
@good-for-nothing-man  Looks like you are not able to load the xclbin. Please make sure xclbin is built properly and correct path of the xclbin is provided.

heeran-xilinx, 2020-12-15:
closing this as no response for last comment. Re-open if issue still exist.

closed 2020-12-15 by heeran-xilinx