Xilinx/Vitis_Accel_Examples #31: Cannot PLRAM example on U280

Hecmay, 2020-12-10:
Hi, 

I got the following error when running PLRAM example in the repo. I got the following error from XRT

```
Device[1]: program successful!
Creating a kernel [test:{test_1}] for CU(1)
[XRT] ERROR: std::bad_alloc
```

My env:
* Vitis 2019.2 + 
* U280 data center card

Any thoughts how to solve the bad alloc issue?

Same question posted on Xilinx forum: https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/Cannot-run-bitstream-using-PLRAM-on-U280/td-p/1184694

virata-xilinx, 2020-12-15:
Hi @Hecmay ,

Did the original plram_access example fail for you? If so, can you please mention the steps followed by you? I will try to reproduce the issue.
I see in the query you mentioned that you referred the example and created a modified version of the same which failed. Did that pass for sw_emu and hw_emu for u280?

heeran-xilinx, 2020-12-22:
As there is already a query posted in Xilinx FOrum so closing this topic here.

closed 2020-12-22 by heeran-xilinx