Xilinx/Vitis_Accel_Examples #33: TEST helloworld demo ERROR

chunxue-ding, 2021-01-20:
Hi， when  i  test helloworld ，There is an error as follow
[ding@B318 hello_world]$ make all TARGET=sw_emu DEVICE=xilinx_u50_xdma_201920_1
mkdir -p ./build_dir.sw_emu.xilinx_u50_xdma_201920_1
v++ -l  -t sw_emu --platform xilinx_u50_xdma_201920_1 --save-temps  -g --temp_dir ./build_dir.sw_emu.xilinx_u50_xdma_201920_1  -o'./build_dir.sw_emu.xilinx_u50_xdma_201920_1/vadd.link.xclbin' _x.sw_emu.xilinx_u50_xdma_201920_1/vadd.xo
Option Map File Used: '/tools/Xilinx/Vitis/2019.2/data/vitis/vpp/optMap.xml'

****** v++ v2019.2 (64-bit)
  **** SW Build 2700185 on Thu Oct 24 18:45:48 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ link can be found at:
	Reports: /hdd/F_C_ding/Vitis_Accel_Examples/hello_world/build_dir.sw_emu.xilinx_u50_xdma_201920_1/reports/link
	Log files: /hdd/F_C_ding/Vitis_Accel_Examples/hello_world/build_dir.sw_emu.xilinx_u50_xdma_201920_1/logs/link
Running Dispatch Server on port:44082
INFO: [v++ 60-1548] Creating build summary session with primary output /hdd/F_C_ding/Vitis_Accel_Examples/hello_world/build_dir.sw_emu.xilinx_u50_xdma_201920_1/vadd.link.xclbin.link_summary, at Wed Jan 20 14:54:38 2021
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Wed Jan 20 14:54:38 2021
Running Rule Check Server on port:40035
INFO: [v++ 60-1315] Creating rulecheck session with output '/hdd/F_C_ding/Vitis_Accel_Examples/hello_world/build_dir.sw_emu.xilinx_u50_xdma_201920_1/reports/link/v++_link_vadd.link_guidance.html', at Wed Jan 20 14:54:39 2021
INFO: [v++ 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u50_xdma_201920_1/xilinx_u50_xdma_201920_1.xpfm
INFO: [v++ 60-1578]   This platform contains Device Support Archive '/opt/xilinx/platforms/xilinx_u50_xdma_201920_1/hw/xilinx_u50_xdma_201920_1.dsa'
INFO: [v++ 60-629] Linking for software emulation target
INFO: [v++ 60-423]   Target device: xilinx_u50_xdma_201920_1
INFO: [v++ 60-645] kernel flags are '-g -I /hdd/F_C_ding/Vitis_Accel_Examples/hello_world/src -g'
INFO: [v++ 60-586] Created ./build_dir.sw_emu.xilinx_u50_xdma_201920_1/vadd.link.xclbin
INFO: [v++ 60-1307] Run completed. Additional information can be found in:
	Guidance: /hdd/F_C_ding/Vitis_Accel_Examples/hello_world/build_dir.sw_emu.xilinx_u50_xdma_201920_1/reports/link/v++_link_vadd.link_guidance.html
	Steps Log File: /hdd/F_C_ding/Vitis_Accel_Examples/hello_world/build_dir.sw_emu.xilinx_u50_xdma_201920_1/logs/link/link.steps.log

INFO: [v++ 60-791] Total elapsed time: 0h 0m 9s
v++ -p ./build_dir.sw_emu.xilinx_u50_xdma_201920_1/vadd.link.xclbin -t sw_emu --platform xilinx_u50_xdma_201920_1 --package.out_dir ./package.sw_emu -o ./build_dir.sw_emu.xilinx_u50_xdma_201920_1/vadd.xclbin
ERROR: [v++ 60-1520] ***Exception: unrecognised option '-p' 
usage: v++ [options] <input file...>

Because it's the first time，hope some advices ,thankyou.

virata-xilinx, 2021-01-20:
Hi @chunxue-ding ,

Can you please mention which branch you cloned and which VITIS and XRT version you have sourced. I guess the issue might be because you cloned the 2020 branch but you are sourcing 2019 VITIS and XRT.
The failure occurs because packaging command was not supported till 2019.

chunxue-ding, 2021-01-20:
yes is indeed , i sourcing 2019 VITIS and XRT ，so should i  sourcing 2020 VITIS and XRT to sove this problem ? i didn't find 2019 branch

virata-xilinx, 2021-01-20:
It is recommended that you source the same VITIS and XRT as the branch. If you are trying with 2020 branch you should source the respective VITIS and XRT (2020.1 for 2020.1 branch and 2020.2 for master branch).
Similarly for 2019.2 branch (https://github.com/Xilinx/Vitis_Accel_Examples/tree/2019.2) you should source 2019.2 VITIS and XRT

chunxue-ding, 2021-01-20:
Thank you for you advices .  I have solved the problem according to your suggestion.

closed 2021-01-20 by chunxue-ding