Xilinx/Vitis_Accel_Examples #34: Xilinx vitis HW emulation exception caught error

dheeraj141, 2021-02-22:
I have been trying to solve this error for over a month but no results every time I run the HW gives this error . the software emulation runs fine but don't know what's the problem with the hardware . the memory also is not that much allocated just a matrix of 32*32. 

[XRT] WARNING: Profiling may contain incomplete information. Please ensure all OpenCL objects are released by your host code (e.g., clReleaseProgram()).
ERROR: [EMU 60-601]  Exception Caught - Failed with the command REMOVE operation at the Line Number: 1394
Makefile:135: recipe for target 'check' failed
make: *** [check] Error 1

vishnuchebrolu, 2021-02-22:
@dheeraj141, Is it possible to share the code so that we can check the failure?
 which Vitis & XRT are being used?

-Vishnu

dheeraj141, 2021-02-22:
It is the hbm bandwidth example but I modified the function performed by the kernel. The kernel is taking 32 inputs each input corresponding to a different bank and then performing a function on the inputs. 

XRT Information
Version:        2.7.766
[0] 0000:05:00.1 xilinx_u280_xdma_201920_3(ID=0x5e278820) user(inst=128)

vishnuchebrolu, 2021-02-23:
Hi @dheeraj141 
As you modified the function performed by the kernel, it reports following errors in log:
[XRT] WARNING: Profiling may contain incomplete information. Please ensure all OpenCL objects are released by your host code (e.g., clReleaseProgram()).
ERROR: [EMU 60-601] Exception Caught - Failed with the command REMOVE operation at the Line Number: 1394
Makefile:135: recipe for target 'check' failed
make: *** [check] Error 1

Please do use Xilinx Forum for such queries:
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v

Please use this Git Issues if you are facing any issue with any existing design without any change in example.

Thanks,
Vishnu

heeran-xilinx, 2021-03-04:
Closing as same discussion started in forum

closed 2021-03-04 by heeran-xilinx