Xilinx/Vitis_Accel_Examples #35: Failed to comile 'host/concurrent_kernel_execution'

LemonAndRabbit, 2021-03-24:
I am a beginner in Vitis. I'm trying the example in 'host/concurrent_kernel_execution'.

My command is as follows:
```bash
make all TARGET=sw_emu DEVICE=xilinx_u50_gen3x16_xdma_201920_3
```

Here is the full response:
```text
g++ -o concurrent_kernel_execution /home/yzf/download/Vitis_Accel_Examples/common/includes/xcl2/xcl2.cpp src/host.cpp -I/opt/xilinx/xrt/include -I/home/yzf/FPGA/Vivado/2020.2/include -Wall -O0 -g -std=c++11 -I/home/yzf/download/Vitis_Accel_Examples/common/includes/xcl2 -fmessage-length=0 -L/opt/xilinx/xrt/lib -lOpenCL -lpthread  -lrt -lstdc++ 
mkdir -p ./_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3
v++ -t hw_emu --platform xilinx_u50_gen3x16_xdma_201920_3 --save-temps  -g -c -k madd --temp_dir ./_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3  -I'src' -o'_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/madd.xo' 'src/madd.cpp'
Option Map File Used: '/home/yzf/FPGA/Vitis/2020.2/data/vitis/vpp/optMap.xml'

****** v++ v2020.2 (64-bit)
  **** SW Build (by xbuild) on 2020-11-18-05:13:29
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ compile can be found at:
        Reports: /home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/reports/madd
        Log files: /home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/logs/madd
Running Dispatch Server on port:41803
INFO: [v++ 60-1548] Creating build summary session with primary output /home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/madd.xo.compile_summary, at Wed Mar 24 16:51:54 2021
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Wed Mar 24 16:51:54 2021
Running Rule Check Server on port:44875
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/reports/madd/v++_compile_madd_guidance.html', at Wed Mar 24 16:51:55 2021
INFO: [v++ 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u50_gen3x16_xdma_201920_3/xilinx_u50_gen3x16_xdma_201920_3.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/opt/xilinx/platforms/xilinx_u50_gen3x16_xdma_201920_3/hw/hw.xsa'
INFO: [v++ 74-74] Compiler Version string: 2020.2
INFO: [v++ 60-585] Compiling for hardware emulation target
INFO: [v++ 60-423]   Target device: xilinx_u50_gen3x16_xdma_201920_3
INFO: [v++ 60-242] Creating kernel: 'madd'
WARNING: /home/yzf/FPGA/Vitis/2020.2/tps/lnx64/jre9.0.4 does not exist.

===>The following messages were generated while  performing high-level synthesis for kernel: madd Log file: /home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/madd/madd/vitis_hls.log :
INFO: [v++ 204-61] Pipelining loop 'madd_readA'.
INFO: [v++ 200-1470] Pipelining result : Target II = 1, Final II = 1, Depth = 3, loop 'madd_readA'
INFO: [v++ 204-61] Pipelining loop 'madd_readB'.
INFO: [v++ 200-1470] Pipelining result : Target II = 1, Final II = 1, Depth = 3, loop 'madd_readB'
INFO: [v++ 204-61] Pipelining loop 'madd_writeC'.
INFO: [v++ 200-1470] Pipelining result : Target II = 1, Final II = 1, Depth = 4, loop 'madd_writeC'
INFO: [v++ 200-790] **** Loop Constraint Status: All loop constraints were satisfied.
INFO: [v++ 200-789] **** Estimated Fmax: 411.00 MHz
INFO: [v++ 60-594] Finished kernel compilation
INFO: [v++ 60-244] Generating system estimate report...
INFO: [v++ 60-1092] Generated system estimate report: /home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/reports/madd/system_estimate_madd.xtxt
INFO: [v++ 60-586] Created _x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/madd.xo
INFO: [v++ 60-2343] Use the vitis_analyzer tool to visualize and navigate the relevant reports. Run the following command. 
    vitis_analyzer /home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/madd.xo.compile_summary 
INFO: [v++ 60-791] Total elapsed time: 0h 0m 42s
INFO: [v++ 60-1653] Closing dispatch client.
mkdir -p ./_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3
v++ -t hw_emu --platform xilinx_u50_gen3x16_xdma_201920_3 --save-temps  -g -c -k mscale --temp_dir ./_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3  -I'src' -o'_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/mscale.xo' 'src/mscale.cpp'
Option Map File Used: '/home/yzf/FPGA/Vitis/2020.2/data/vitis/vpp/optMap.xml'

****** v++ v2020.2 (64-bit)
  **** SW Build (by xbuild) on 2020-11-18-05:13:29
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ compile can be found at:
        Reports: /home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/reports/mscale
        Log files: /home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/logs/mscale
Running Dispatch Server on port:45013
INFO: [v++ 60-1548] Creating build summary session with primary output /home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/mscale.xo.compile_summary, at Wed Mar 24 16:52:39 2021
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Wed Mar 24 16:52:39 2021
Running Rule Check Server on port:41989
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/reports/mscale/v++_compile_mscale_guidance.html', at Wed Mar 24 16:52:40 2021
INFO: [v++ 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u50_gen3x16_xdma_201920_3/xilinx_u50_gen3x16_xdma_201920_3.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/opt/xilinx/platforms/xilinx_u50_gen3x16_xdma_201920_3/hw/hw.xsa'
INFO: [v++ 74-74] Compiler Version string: 2020.2
[... 89 lines cut ...]
error: file /home/yzf/%E4%B8%8B%E8%BD%BD/Vitis_Accel_Examples/host/concurrent_kernel_execution/build_dir.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/link/sys_link/iprepo/temp/xo0/madd/kernel.xml
xsltRunStylesheet : run failed
ERROR: [SYSTEM_LINK 82-63] Error generating intermediate file /home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/build_dir.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/link/sys_link/iprepo/temp/xo0/ip_repo/xilinx_com_hls_madd_1_0/madd.fcnmap.xml
ERROR: [SYSTEM_LINK 82-78] Unable to create function map for HLS kernel madd
ERROR: [SYSTEM_LINK 82-84] Unable to process .xo file: /home/yzf/download/Vitis_Accel_Examples/host/concurrent_kernel_execution/_x.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/madd.xo
ERROR: [SYSTEM_LINK 82-66] Error processing .xo files
ERROR: [SYSTEM_LINK 82-100] Error processing object files, exiting
INFO: [v++ 60-1442] [16:54:15] Run run_link: Step system_link: Failed
Time (s): cpu = 00:00:02 ; elapsed = 00:00:02 . Memory (MB): peak = 1580.469 ; gain = 0.000 ; free physical = 3930 ; free virtual = 14150
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [Makefile:150：build_dir.hw_emu.xilinx_u50_gen3x16_xdma_201920_3/matrix_ops.xclbin] Error 1

```

I don't understand why there is such an error:
```text
ERROR: [KernelCheck 83-117] 'madd' kernel.xml indicates the existence of port S_AXI_CONTROL, which does not exist in component.xml
```

I meet the same error when trying hw_emu in 'host/streaming_k2k_mm', while sw_emu in both example works fine. How should I deal with it?

Thanks a lot for any suggestion~

virata-xilinx, 2021-03-24:
Hi @LemonAndRabbit ,

Can you please mention the branch you have cloned and the VITIS and XRT you have sourced.

LemonAndRabbit, 2021-03-24:
Hello

I'm using the current master branch of this repo:
```text
commit 22855e05235d7f48ebcc708a0dfcadc08af8cc99 (HEAD -> master, origin/master, origin/HEAD)
Author: Virat Agarwal <virat.agarwal@xilinx.com>
Date:   Tue Mar 16 09:14:00 2021 +0530

    Updating sys_opt/multiple_devices example to have larger buffers
```

The VITIS is 2020.2 on Ubuntu 20.04.

The XRT's version.json file is as follows:
```json
{
  "BUILD_VERSION" : "2.8.743",
  "BUILD_VERSION_DATE" : "Mon, 16 Nov 2020 00:18:20 -0800",
  "BUILD_BRANCH" : "2020.2",
  "VERSION_HASH" : "77d5484b5c4daa691a7f78235053fb036829b1e9",
  "VERSION_HASH_DATE" : "Fri, 13 Nov 2020 07:51:30 -0800"
}
```

virata-xilinx, 2021-03-24:
Hi @LemonAndRabbit ,

I am not able to reproduce the issue (The example is passing for me). Can you please mention the exact steps you followed?
Also can you once try the following command once -
make run TARGET=sw_emu DEVICE=/proj/xbuilds/2020.2_daily_latest/internal_platforms/xilinx_u50_gen3x16_xdma_201920_3/xilinx_u50_gen3x16_xdma_201920_3.xpfm

LemonAndRabbit, 2021-03-24:
I made no change to any file, the only thing I do is to type in the command:
```bash
make all TARGET=hw_emu DEVICE=xilinx_u50_gen3x16_xdma_201920_3
```
in directory 'host/concurrent_kernel_execution'.

And the command:
```bash
make run TARGET=sw_emu DEVICE=xilinx_u50_gen3x16_xdma_201920_3
```
get the reponse 'TEST PASSED'.

I made a mistake that I mean the hw_emu cannot pass in my compter, not sw_emu. The sw_emu works just fine, I have edited the problem discription.

virata-xilinx, 2021-03-24:
Hi @LemonAndRabbit,

I ran the example even for hw_emu and the same is working for me. I sourced 2020.2_released VITIS and XRT (same as yours) and ran the following command -
make run TARGET=hw_emu DEVICE=/proj/xbuilds/2020.2_released/internal_platforms/xilinx_u50_gen3x16_xdma_201920_3/xilinx_u50_gen3x16_xdma_201920_3.xpfm
Can you please do a make cleanall once and retry using the above command?

virata-xilinx, 2021-03-24:
Changes made for 1 example -
https://gitenterprise.xilinx.com/SDxGitExamples/Vitis_Accel_Examples/pull/761

LemonAndRabbit, 2021-03-24:
Thanks a lot, I successfully run hw_emu after moving the whole project to a path containing no Chinese characters.

closed 2021-03-24 by LemonAndRabbit