Xilinx/Vitis_Accel_Examples #36: Streaming Example ( K2K) for Xilinx Vitis

dheeraj141, 2021-04-12:
Is there an example similar to streaming_k2k_mm for the cpp kernels because Xilinx Vitis shows error while running the same example ( as it is ) . The following error comes 

ERROR: [v++ 17-1309] Gcc: /tools/reconfig/xilinx/Vitis/2020.1/Vitis/2020.1/bin/../data/emulation/include/xcl_top_defines.h:57:request for member ‘empty’ in ‘* rStream’, which is of non-class type ‘int’
ERROR: [v++ 17-1309] Gcc: /tools/reconfig/xilinx/Vitis/2020.1/Vitis/2020.1/bin/../data/emulation/include/xcl_top_defines.h:58:request for member ‘read’ in ‘* rStream’, which is of non-class type ‘int’

virata-xilinx, 2021-04-13:
Hi @dheeraj141 ,

Can you please mention which flow and platform you are running the example for. Can you please also mention the repo you cloned and the steps you followed when getting the error. I will try to reproduce the issue.
Can you please elaborate on "Is there an example similar to streaming_k2k_mm for the cpp kernels" as well.

heeran-xilinx, 2021-04-20:
Closing this as no response to follow-up question. Please re-open with more details.

closed 2021-04-20 by heeran-xilinx