Xilinx/Vitis_Accel_Examples #37: synth ERROR in U50 AAT design

saimt2019507, 2021-04-15:
Hi,

I use vitis, vivado 2020.2 for U50 accelerator, provided by xilinx.

INFO: [v++ 60-1454] Run Directory: /home/iiitb/Accelerated_Algorithmic_Trading/build/_x/link/run_link

****** vpl v2020.2 (64-bit)
**** SW Build (by xbuild) on 2020-11-18-05:13:29
** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [VPL 60-839] Read in kernel information from file '/home/iiitb/Accelerated_Algorithmic_Trading/build/_x/link/int/kernel_info.dat'.
INFO: [VPL 74-74] Compiler Version string: 2020.2
INFO: [VPL 60-423] Target device: xilinx_u50_gen3x16_xdma_201920_3
INFO: [VPL 60-1032] Extracting hardware platform to /home/iiitb/Accelerated_Algorithmic_Trading/build/_x/link/vivado/vpl/.local/hw_platform
WARNING: /tools/Vitis/Vitis/2020.2/tps/lnx64/jre9.0.4 does not exist.
[13:02:53] Run vpl: Step create_project: Started
Creating Vivado project.
[13:02:58] Run vpl: Step create_project: Completed
[13:02:58] Run vpl: Step create_bd: Started
[13:04:15] Run vpl: Step create_bd: RUNNING...
[13:05:06] Run vpl: Step create_bd: Completed
[13:05:06] Run vpl: Step update_bd: Started
[13:05:08] Run vpl: Step update_bd: Completed
[13:05:08] Run vpl: Step generate_target: Started
[13:06:23] Run vpl: Step generate_target: RUNNING...
[13:07:39] Run vpl: Step generate_target: RUNNING...
[13:08:54] Run vpl: Step generate_target: RUNNING...
[13:10:09] Run vpl: Step generate_target: RUNNING...
[13:10:35] Run vpl: Step generate_target: Completed
[13:10:35] Run vpl: Step config_hw_runs: Started
[13:10:41] Run vpl: Step config_hw_runs: Completed
[13:10:41] Run vpl: Step synth: Started
[13:11:43] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:12:17] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:12:48] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:13:19] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:13:49] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:14:20] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:14:51] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:15:22] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:15:55] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:16:28] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:20:07] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:20:09] Block-level synthesis in progress, 0 of 12 jobs complete, 4 jobs running.
[13:20:09] Run vpl: FINISHED. Run Status: synth ERROR
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
WARNING: [VPL 60-1142] Unable to read data from '/home/iiitb/Accelerated_Algorithmic_Trading/build/_x/link/vivado/vpl/output/generated_reports.log', generated reports will not be copied.
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [13:20:12] Run run_link: Step vpl: Failed
Time (s): cpu = 00:07:52 ; elapsed = 00:17:53 . Memory (MB): peak = 1577.230 ; gain = 0.000 ; free physical = 149 ; free virtual = 2751
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [Makefile:206: aat.xclbin] Error 1

Can you please tell me how to solve this problem?

I am attaching the log file.

Thanks in Advance

virata-xilinx, 2021-04-15:
Hi @saimt2019507 ,

Can you please confirm if you are running a Vitis Accel Example? 
If yes, can you please mention the example and the steps you followed. I will try to reproduce the issue. 
If not, you can post your query on the Xilinx forum.

saimt2019507, 2021-04-18:
Hai,
I am running the example given by Xilinx. I am doing the work related to my
job. Our company bought U50 board. I followed the steps in the manual.
Thank you

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
18/04/21,
03:29:26 pm

heeran-xilinx, 2021-04-19:
Please post your query to Xilinx Forum below:
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v
This Git Repo is specific to Examples provided under this repo.

closed 2021-04-19 by heeran-xilinx