Xilinx/Vitis_Accel_Examples #39: folder path

KarimH22, 2021-05-21:
Hello
is it possible please to put the common path into another repo please ? and so to use it in Vitis_Accel_Examples as a git submodule ?
So that every one will not have to clone all examples  when they need to create their own project 
it could be in Vitis_Accel_Common repo

@BhaskarVishnu
 
thank you

heeran-xilinx, 2021-06-01:
Hi @KarimH22 ,
Thanks for your feedback. We do not want to maintain two separate repos as it brings extra overhead of synchronization, so for sake of simplicity we kept single repo. And if any user is interested to re-use anything from repo, kindly clone full repo.
-Heera

KarimH22, 2021-06-03:
Hi @heeran-xilinx 
thank you for your answer.
Even for the path Vitis_Accel_Examples/common/includes
it seems to be a stable path not so much changed .. and so few synchro.

Best Regards
Karim

heeran-xilinx, 2021-06-08:
Hi @KarimH22 ,
Yes, these are mostly stable path. But having separate repos just for common still not feel good. Anyway we will review this request internally. 
I think you can clone entire repo and refer only common section for your purpose. For faster clone and reduce side,  you can use "git clone --depth 1".

KarimH22, 2021-06-14:
Hi @heeran-xilinx 
ok I will do as you advised , thank you.

closed 2021-06-14 by KarimH22