Xilinx/Vitis_Accel_Examples #42: Something About data_transfer

pyhundan, 2021-07-20:
Hello,

When I was running the test code DATA_TRANSFER, I wanted to change the read file to another place and change the size of the read file. I found that the size created in the test example was:
`204K	dummy_kernel.xclbin`
What should I do if I want to create the file somewhere else and change the file size?

Thanks,
Hundan

virata-xilinx, 2021-07-21:
Hi @pyhundan ,

The xclbin is generated after kernel building. We don't have a control on its size.
However, you can control where you wish to put the xclbin by modifying the path of the "BUILD_DIR" variable appropriately in the Makefile -
https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/host/data_transfer/Makefile#L71

pyhundan, 2021-07-22:
Hi, @virata-xilinx 
Thank you for your answer, but I want to change the https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/host/data_transfer/src/host.cpp#L70
size of the "binaryFile" to test the data transfer under different file sizes. Can I change this size?

virata-xilinx, 2021-07-22:
Hi @pyhundan,

Can you please tell me why are trying to experiment with the xclbin sizes? Are you trying to experiment with the buffer sizes instead? If so you can change the number of elements in the host code -
https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/host/data_transfer/src/host.cpp#L25

pyhundan, 2021-07-22:
Hi @virata-xilinx ,
I just want to understand the different sizes of data transferred from disk to FPGA. So I am going to change the size of the read file to run this data_transfer test code. If this .xclbin is not the read file size, how should I write a code?

vishnuchebrolu, 2021-07-22:
Hi @pyhundan ,

Are you looking for the following design : 
https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/host/host_global_bandwidth

Thanks,
Vishnu

pyhundan, 2021-07-22:
Hi @vishnuchebrolu ,
When I run this sample code, I get an error.
`/usr/include/CL/cl2.hpp:7841:30: error: expected ‘;’ at end of member declaration
         Event* event = NULL) CL_EXT_SUFFIX__VERSION_1_2_DEPRECATED const
                              ^
In file included from /home/fpgauser/fpga_code/workpalce/test/libs/common/includes/xcl2/xcl2.hpp:33:0,
                 from ../src/host.cpp:24:
/usr/include/CL/cl2.hpp:7842:5: error: expected unqualified-id before ‘{’ token
     {
~                        `
My cl2.hpp file mentions version 2.0.7. May this be due to a mismatch in our OpenCL versions?

virata-xilinx, 2021-07-23:
Hi @pyhundan ,

Can you please mention the VITIS and XRT sourced before running the code? Also can you please mention the steps you followed? I will try to reproduce the issue.

pyhundan, 2021-07-25:
Hi @virata-xilinx ，
The code is https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/host/host_global_bandwidth 
The problem was observed xilinx_u200_xdma_201830_2 platform, and with XRT 2020.2.
The Host system was Ubuntu 16.04. The Vitis IDE's version is v2020.2.0(64bit).

virata-xilinx, 2021-07-25:
Hi @pyhundan ,

The error message you observe seems to be occurring at the host compilation stage. However, I couldn't reproduce the issue locally with the settings you mentioned (master, XRT 2020.2, u200_xdma_201830_2).
I would also like to point out that the master branch is supposed to be run with 2021.1 XRT (and the 2020.2 branch with 2020.2 XRT). Can you please try with that once?
As you mentioned, the issue might also be occurring due to the cl2.hpp version (Mine was 2.0.10). Can you please fetch the latest and try again once? Can you please also tell me if the "data_transfer" example was also failing for you with the same error?

heeran-xilinx, 2021-08-03:
Closing as no respond so far. Please re-open if still facing the issue.
For better Vitis related queries, you can directly post on Xilinx Forum:

https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v

closed 2021-08-03 by heeran-xilinx