Xilinx/Vitis_Accel_Examples #43: Errors when running software emulation on Vitis_Accel_Examples/host_xrt/hbm_simple_xrt

jchen8tw, 2021-08-05:
Building the code with my Vitis IDE using software emulation gives no error.
But running it gives the error message in the following.
I use all the default building configuration from this repo and Vitis IDE.
Please help me out. Thanks a lot.

```
Open the device0
Load the xclbin /home/edci/vitis_workspace/HBM_XRT_system/Emulation-SW/krnl_vadd.xclbin
Running CASE 1  : Single HBM for all three Buffers 
input 1 -> bank 0 
input 2 -> bank 0 
output  -> bank 0 
Allocate Buffer in Global Memory
synchronize input buffer data to device global memory
Execution of the kernel
Get the output data from the device
[CASE 1] THROUGHPUT = 1.68053 GB/s
Running CASE 2: Three Separate Banks for Three Buffers
input 1 -> bank 1 
input 2 -> bank 2 
output  -> bank 3 
Allocate Buffer in Global Memory
synchronize input buffer data to device global memory
Execution of the kernel
terminate called after throwing an instance of 'std::runtime_error'
  what():  No compute units satisfy requested connectivity
```

virata-xilinx, 2021-08-05:
Hi @jchen8tw ,

Thanks for reporting the issue. The example is actually disabled for sw_emu as can be observed in the description.json file -
https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/host_xrt/hbm_simple_xrt/description.json#L33
The same is however not reflected in the IDE and thus the example runs and throws an error.
You may try to run the example for any other flow on the IDE.

jchen8tw, 2021-08-06:
@virata-xilinx Thank you for your kindly replay!

closed 2021-08-06 by jchen8tw