Xilinx/Vitis_Accel_Examples #46: Multiple versions of xcl2.hpp

ahonorat, 2021-09-30:
Hello,
I am using the functions defined in xcl2.hpp and xcl2.cpp files in my host code for OpenCL accelerated kernel on a Xilinx platform.
However, I have noticed that multiple different versions of this file exist in this repository and also in the one for Vitis Libraries :
https://github.com/Xilinx/Vitis_Libraries
I am not sure that this is a real issue, but it seems weird enough to be asked.

**Question**: Which version of the header should I use? I also noticed that there is a dependency to "CL/cl_ext_xilinx.h" in some headers, so is "xcl2.hpp" platform- or version-dependent?

Here are the sha1sum of all xcl2.hpp found in this repository and in the one for Vitis Libraries :
```
[Vitis_Accel_Examples]find . -name "xcl2.hpp" | xargs -I {} sha1sum {}

3d923b533c2cab84fcee807dcda0cd290fc89fa1  ./library_examples/gemm/include/sw/xcl2/xcl2.hpp
71acdaeb2de97a706eeb827e4be2f60e2fdb5961  ./library_examples/MCEuropeanEngine/src/xcl2.hpp
35ee2824550aeead5ac1c8eec3e5d29a78490e08  ./library_examples/anti_money_laundering/host/xcl2.hpp
fd288f83104689f39542c9c23668fb9cfc5fef8d  ./library_examples/gzip_app/include/xcl2.hpp
964b7ad69b305ec1a825bb17e0df806b8903b888  ./common/includes/xcl2/xcl2.hpp

[Vitis_Libraries]find . -name "xcl2.hpp" | xargs -I {} sha1sum {}

38a7f4189b77621ae4dde50e1d27dd1f9fe04121  ./codec/ext/xcl2/xcl2.hpp
38a7f4189b77621ae4dde50e1d27dd1f9fe04121  ./database/ext/xcl2/xcl2.hpp
38a7f4189b77621ae4dde50e1d27dd1f9fe04121  ./dsp/ext/xcl2/xcl2.hpp
38a7f4189b77621ae4dde50e1d27dd1f9fe04121  ./utils/ext/xcl2/xcl2.hpp
38a7f4189b77621ae4dde50e1d27dd1f9fe04121  ./graph/ext/xcl2/xcl2.hpp
38a7f4189b77621ae4dde50e1d27dd1f9fe04121  ./security/ext/xcl2/xcl2.hpp
38a7f4189b77621ae4dde50e1d27dd1f9fe04121  ./solver/ext/xcl2/xcl2.hpp

c041215de4ef2a9012f78417901addeeb398f8ec  ./blas/L2/include/xcl2/xcl2.hpp
ab1b6c94039ae4b7d84361e48ec78e22c0c4f636  ./data_analytics/ext/xcl2/xcl2.hpp
8faace4fddb78a7a364d972bd13abb9a1ba7a8fc  ./data_compression/common/libs/xcl2/xcl2.hpp
508a1771aa3ed36c6b8bc0d2ae248e063ffa394c  ./utils/L2/tests/utils_sw/shared_host/xcl2/xcl2.hpp
7e6925513718737175c5413c58a9efde1bf86131  ./quantitative_finance/ext/xcl2/xcl2.hpp
3959e63c54da8831968f58d210f27f8bf69c9c4d  ./vision/ext/xcl2/xcl2.hpp
```
(I've reordered the output to cluster the similar sha1 together, they are from yesterday in the main branch).

ahonorat, 2021-09-30:
See crossed issue in the repository of Vitis Libraries:
https://github.com/Xilinx/Vitis_Libraries/issues/84

heeran-xilinx, 2021-09-30:
HI @ahonorat , 
xcl2.hpp is a helper code and every repository is modifying it for thier own convenience and need basis. As a result, you could 
 see difference across repositories.  So I would recommend, please use based on your requirement.

heeran-xilinx, 2021-09-30:
[Vitis_Accel_Examples]find . -name "xcl2.hpp" | xargs -I {} sha1sum {}

3d923b533c2cab84fcee807dcda0cd290fc89fa1  ./library_examples/gemm/include/sw/xcl2/xcl2.hpp
71acdaeb2de97a706eeb827e4be2f60e2fdb5961  ./library_examples/MCEuropeanEngine/src/xcl2.hpp
35ee2824550aeead5ac1c8eec3e5d29a78490e08  ./library_examples/anti_money_laundering/host/xcl2.hpp
fd288f83104689f39542c9c23668fb9cfc5fef8d  ./library_examples/gzip_app/include/xcl2.hpp
964b7ad69b305ec1a825bb17e0df806b8903b888  ./common/includes/xcl2/xcl2.hpp
Designs under library_examples are few designs which was imported from Library. So they have thier own version xcl2.hpp which were added here.

ahonorat, 2021-09-30:
Hi @heeran-xilinx , thanks for your quic reply!
> So I would recommend, please use based on your requirement.

Do you know the differences between them? Which one is the most standard/up-to-date one?
I don't need anything specific: just to load an xclbin, create corresponding OpenCL kernels and launch them in the processing queue.

heeran-xilinx, 2021-09-30:
Hi @ahonorat , 

For latest, please refer below:
https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/common/includes/xcl2

-Heera

ahonorat, 2021-10-01:
Ok thanks, closing the issue then.

closed 2021-10-01 by ahonorat