Xilinx/Vitis_Accel_Examples #51: errors on "kmeans example"

lloo099, 2021-11-18:
```
 make all TARGET=sw_emu PLATFORM=xilinx_u200_xdma_201830_2 HOST_ARCH=x86
g++ -o kmeans /home/enai/Desktop/project/enai0/Vitis_Accel_Examples/common/includes/xcl2/xcl2.cpp /home/enai/Desktop/project/enai0/Vitis_Accel_Examples/common/includes/cmdparser/cmdlineparser.cpp /home/enai/Desktop/project/enai0/Vitis_Accel_Examples/common/includes/logger/logger.cpp src/host.cpp src/fpga_kmeans.cpp src/kmeans_clustering_cmodel.c -D __USE_OPENCL__ -DNUM_CU=2 -I/opt/xilinx/xrt/include -I/tools/Xilinx/Vivado/2020.1/include -Wall -O0 -g -std=c++1y -I/home/enai/Desktop/project/enai0/Vitis_Accel_Examples/common/includes/xcl2 -I/home/enai/Desktop/project/enai0/Vitis_Accel_Examples/common/includes/cmdparser -I/home/enai/Desktop/project/enai0/Vitis_Accel_Examples/common/includes/logger -fmessage-length=0 -L/opt/xilinx/xrt/lib -lOpenCL -pthread -lrt -lstdc++ 
In file included from /home/enai/Desktop/project/enai0/Vitis_Accel_Examples/common/includes/xcl2/xcl2.cpp:17:0:
/home/enai/Desktop/project/enai0/Vitis_Accel_Examples/common/includes/xcl2/xcl2.hpp:33:10: fatal error: CL/cl2.hpp: No such file or directory
 #include <CL/cl2.hpp>
          ^~~~~~~~~~~~
compilation terminated.
In file included from src/fpga_kmeans.h:28:0,
                 from src/host.cpp:56:
/home/enai/Desktop/project/enai0/Vitis_Accel_Examples/common/includes/xcl2/xcl2.hpp:33:10: fatal error: CL/cl2.hpp: No such file or directory
 #include <CL/cl2.hpp>
          ^~~~~~~~~~~~
compilation terminated.
In file included from src/fpga_kmeans.h:28:0,
                 from src/fpga_kmeans.cpp:17:
/home/enai/Desktop/project/enai0/Vitis_Accel_Examples/common/includes/xcl2/xcl2.hpp:33:10: fatal error: CL/cl2.hpp: No such file or directory
 #include <CL/cl2.hpp>
          ^~~~~~~~~~~~
compilation terminated.
Makefile:158: recipe for target 'kmeans' failed
make: *** [kmeans] Error 1

```

virata-xilinx, 2021-11-18:
Hi @lloo099 ,

Can you please metion the branch you have cloned. Kindly also mention the VITIS and XRT you have sourced and the steps to reproduce.

lloo099, 2021-11-18:
> Hi @lloo099 ,
> 
> Can you please metion the branch you have cloned. Kindly also mention the VITIS and XRT you have sourced and the steps to reproduce.

Sure @virata-xilinx , I did it follows these steps:
```
git clone https://github.com/Xilinx/Vitis_Accel_Examples.git
export VIVADO_PATH=/tools/Xilinx/Vivado/2020.1
source /tools/Xilinx/Vitis/2020.1/settings64.sh
source /opt/xilinx/xrt/setup.sh
make all TARGET=sw_emu PLATFORM=xilinx_u200_xdma_201830_2 HOST_ARCH=x86
```

virata-xilinx, 2021-11-18:
@lloo099 , So you cloned the master branch and are sourcing 2020.1? For master you should be sourcing 2021.2. If you wish to source 2020.1, kindly clone the 2020.1 branch.

lloo099, 2021-11-18:
> @lloo099 , So you cloned the master branch and are sourcing 2020.1? For master you should be sourcing 2021.2. If you wish to source 2020.1, kindly clone the 2020.1 branch.

Yes, wow. How to clone 2020.1 branch? Could you give me a link? thanks

virata-xilinx, 2021-11-18:
Kindly use the following link with the "git clone" command to clone the 2020.1 branch - 
https://github.com/Xilinx/Vitis_Accel_Examples/tree/2020.1

lloo099, 2021-11-18:
> Kindly use the following link with the "git clone" command to clone the 2020.1 branch - https://github.com/Xilinx/Vitis_Accel_Examples/tree/2020.1
same problem on 2020.1
```
g++ -o hello_world /home/enai/Desktop/project/enai0/Vitis_Accel_Examples-2020.1/common/includes/xcl2/xcl2.cpp src/host.cpp -I/opt/xilinx/xrt/include -I/tools/Xilinx/Vivado/2020.1/include -Wall -O0 -g -std=c++11 -I/home/enai/Desktop/project/enai0/Vitis_Accel_Examples-2020.1/common/includes/xcl2 -fmessage-length=0 -L/opt/xilinx/xrt/lib -lOpenCL -lpthread  -lrt -lstdc++ 
In file included from /home/enai/Desktop/project/enai0/Vitis_Accel_Examples-2020.1/common/includes/xcl2/xcl2.cpp:17:0:
/home/enai/Desktop/project/enai0/Vitis_Accel_Examples-2020.1/common/includes/xcl2/xcl2.hpp:34:10: fatal error: CL/cl2.hpp: No such file or directory
 #include <CL/cl2.hpp>
          ^~~~~~~~~~~~
compilation terminated.
In file included from src/host.cpp:17:0:
/home/enai/Desktop/project/enai0/Vitis_Accel_Examples-2020.1/common/includes/xcl2/xcl2.hpp:34:10: fatal error: CL/cl2.hpp: No such file or directory
 #include <CL/cl2.hpp>
          ^~~~~~~~~~~~
compilation terminated.

```

keryell, 2021-11-18:
It is not clear what is the system you are using.
On my Debian system the `/usr/include/CL/cl2.hpp` header comes from the `opencl-clhpp-headers` package.
If you are using such a system, you could try an `apt install opencl-clhpp-headers`.

lloo099, 2021-11-19:
> It is not clear what is the system you are using. On my Debian system the `/usr/include/CL/cl2.hpp` header comes from the `opencl-clhpp-headers` package. If you are using such a system, you could try an `apt install opencl-clhpp-headers`.

Thanks, this problem is solved. But new error about platform. Could u help me with it again?
```

make all TARGET=hw_emu DEVICE=xilinx_u200_xdma_201830_2 HOST_ARCH=x86
mkdir -p ./_x.hw_emu.xilinx_u200_xdma_201830_2
v++ -c -k vadd -t hw_emu --platform xilinx_u200_xdma_201830_2 --save-temps  -g --temp_dir ./_x.hw_emu.xilinx_u200_xdma_201830_2  -I'src' -o'_x.hw_emu.xilinx_u200_xdma_201830_2/vadd.xo' 'src/vadd.cpp'
Option Map File Used: '/tools/Xilinx/Vitis/2020.1/data/vitis/vpp/optMap.xml'

****** v++ v2020.1 (64-bit)
  **** SW Build 2902540 on Wed May 27 19:54:35 MDT 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ compile can be found at:
	Reports: /home/enai/Desktop/project/enai0/Vitis_Accel_Examples-2020.1/hello_world/_x.hw_emu.xilinx_u200_xdma_201830_2/reports/vadd
	Log files: /home/enai/Desktop/project/enai0/Vitis_Accel_Examples-2020.1/hello_world/_x.hw_emu.xilinx_u200_xdma_201830_2/logs/vadd
INFO: [v++ 60-1657] Initializing dispatch client.
Running Dispatch Server on port:39199
INFO: [v++ 60-1548] Creating build summary session with primary output /home/enai/Desktop/project/enai0/Vitis_Accel_Examples-2020.1/hello_world/_x.hw_emu.xilinx_u200_xdma_201830_2/vadd.xo.compile_summary, at Fri Nov 19 13:05:10 2021
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Fri Nov 19 13:05:10 2021
Running Rule Check Server on port:40129
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/enai/Desktop/project/enai0/Vitis_Accel_Examples-2020.1/hello_world/_x.hw_emu.xilinx_u200_xdma_201830_2/reports/vadd/v++_compile_vadd_guidance.html', at Fri Nov 19 13:05:12 2021
ERROR: [v++ 60-1258] No valid platform was found that matches 'xilinx_u200_xdma_201830_2'. Please make sure that the platform is specified correctly, and the platform has the right version number. The platform repo paths are:
	/tools/Xilinx/Vitis/2020.1/platforms
	/opt/xilinx/platforms
The valid platforms found from the above repo paths are:

ERROR: [v++ 60-587] Failed to add a platform: specified platform xilinx_u200_xdma_201830_2 is not found or is not valid
ERROR: [v++ 60-592] Failed to finish compilation
INFO: [v++ 60-1653] Closing dispatch client.
Makefile:117: recipe for target '_x.hw_emu.xilinx_u200_xdma_201830_2/vadd.xo' failed
make: *** [_x.hw_emu.xilinx_u200_xdma_201830_2/vadd.xo] Error 1

```

closed 2021-11-19 by lloo099