Xilinx/Vitis_Accel_Examples #53: Error when running 'Hello_world' on U50

IskandarZhang, 2021-12-01:
Hi, after compiling for Alveo U50 Accelerator Card by coding `make all DEVICE=xilinx_u50_gen3x16_xdma_201920_3 TARGET=hw` command the `vadd.xclbin` was generated in fold `build_dir.hw.xilinx_u50_gen3x16_xdma_201920_3`. When I code `./hello_world '/home/username/Vitis_Accel_Examples-master/hello_world/build_dir.hw.xilinx_u50_gen3x16_xdma_201920_3/vadd.xclbin'` some error occured and there was no output. Here follows the error info:
```
XRT build version: 2.11.634
Build hash: 5ad5998d67080f00bca5bf15b3838cf35e0a7b26
Build date: 2021-06-08 22:08:45
Git branch: 2021.1
PID: 1868037
UID: 1008
[Wed Dec  1 01:16:16 2021 GMT]
HOST: server-94
EXE: /home/username/Vitis_Accel_Examples-master/hello_world/hello_world
[XRT] WARNING: "profile" configuration in xrt.ini will be deprecated in the next release.  Please use "opencl_summary=true" to enable OpenCL profiling and "opencl_device_counter=true" for device counter data in OpenCL profile summary.
Found Platform
Platform Name: Xilinx
INFO: Reading /home/username/Vitis_Accel_Examples-master/hello_world/build_dir.hw.xilinx_u50_gen3x16_xdma_201920_3/vadd.xclbin
Loading: '/home/username/Vitis_Accel_Examples-master/hello_world/build_dir.hw.xilinx_u50_gen3x16_xdma_201920_3/vadd.xclbin'
Trying to program device[0]: xilinx_u50_gen3x16_xdma_201920_3
Device[0]: program successful!
terminate called recursively
terminate called after throwing an instance of '__gnu_cxx::recursive_init_error'
Aborted (core dumped)
``` 
Looking forward to your reply.

vishnuchebrolu, 2021-12-01:
Hi @IskandarZhang ,
Can you please provide more details on the Vitis , XRT version and OS information that is being used for running the design.

Try regenerating the host executable using "make host" command on the target HW machine and check again.

Thanks,
Vishnu

IskandarZhang, 2021-12-01:
> Hi @IskandarZhang , Can you please provide more details on the Vitis , XRT version and OS information that is being used for running the design.
> 
> Try regenerating the host executable using "make host" command on the target HW machine and check again.
> 
> Thanks, Vishnu

Hi Vishnu, thanks for your prompt reply. My Vitis version is 2021.1, XRT build version is 2.11.634, Linux version is 20.04.

When I type command `make host` according to Makefile, the output is `make: Nothing to be done for 'host'.` but there is `hello_world` executable file in root directory. And when I code `./hello_world '/home/username/Vitis_Accel_Examples-master/hello_world/build_dir.hw.xilinx_u50_gen3x16_xdma_201920_3/vadd.xclbin'` the same error like before occured.

Best.

vishnuchebrolu, 2021-12-01:
Hi @IskandarZhang ,

I hope you are using 2021.1 branch of repository as you are using 2021.1 Vitis and XRT. In case if you are not, follow the below command to pick the 2021.1 branch.
"git clone https://github.com/Xilinx/Vitis_Accel_Examples.git -b 2021.1"

Can you try running "dmesg" on the machine and share the log output for us to debug.

Thanks,
Vishnu

IskandarZhang, 2021-12-01:
> Hi @IskandarZhang ,
>
> I hope you are using 2021.1 branch of repository as you are using 2021.1 Vitis and XRT. In case if you are not, follow the below command to pick the 2021.1 branch. "git clone [...]

Thanks for your guiding to get the newest clone but I am just using this branch actually. And here follows the output of `dmesg` command
```
[1879234.153324] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 kds_add_context: Client pid(1868098) add context CU(0xffffffff) shared(true)
[1879234.153334] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 kds_del_context: Client pid(1868098) del context CU(0xffffffff)
[1879234.153340] xocl 0000:87:00.1: icap.u.23068674 ffff9ebaef434010 icap_unlock_bitstream: bitstream bb4375fc-b055-23c4-3754-c587632806a2 unlocked, ref=0
[1879234.153700] xocl 0000:87:00.1: icap.u.23068674 ffff9ebaef434010 icap_lock_bitstream: bitstream bb4375fc-b055-23c4-3754-c587632806a2 locked, ref=1
[1879234.153707] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 kds_add_context: Client pid(1868098) add context CU(0xffffffff) shared(true)
[1879234.160671] xocl 0000:87:00.1: icap.u.23068674 ffff9ebaef434010 icap_cached_ocl_frequency: no cached data for 3
[1879234.161666] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 kds_add_context: Client pid(1868098) add context CU(0xffffffff) shared(true)
[1879234.161711] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 kds_add_context: Client pid(1868098) add context CU(0x0) shared(true)
[1879234.330838] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 _kds_fini_client: Client pid(1868098) has 3 opening context
[1879234.330841] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 kds_del_context: Client pid(1868098) del context CU(0xffffffff)
[1879234.330843] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 kds_del_context: Client pid(1868098) del context CU(0xffffffff)
[1879234.330849] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 kds_del_context: Client pid(1868098) del context CU(0x0)
[1879234.330863] xocl 0000:87:00.1: icap.u.23068674 ffff9ebaef434010 icap_unlock_bitstream: bitstream bb4375fc-b055-23c4-3754-c587632806a2 unlocked, ref=0
[1879234.330869] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 xocl_destroy_client: client exits pid(1868098)
[1879234.330873] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 xocl_drvinst_close: CLOSE 2
[1879234.330877] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 xocl_drvinst_close: NOTIFY 0000000073a5a7df
[1879235.876370] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 _xocl_drvinst_open: OPEN 1
[1879235.876414] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 xocl_create_client: created KDS client for pid(1868118), ret: 0
[1879235.878898] xocl 0000:87:00.1: xmc.u.19922946 ffff9ebaef430010 xmc_read_from_peer: reading from peer
[1879235.878919] xocl 0000:87:00.1: mailbox.u.9437186 ffff9ebbf58f6c10 mailbox_request: sending request: 10 via HW
[1879235.881203] xclmgmt 0000:87:00.0: mailbox.m.9437186 ffff9ebb67bf2010 process_request: received request from peer: 10, passed on
[1879235.881214] xclmgmt 0000:87:00.0: xclmgmt_read_subdev_req: req kind 0
[1879235.881296] xclmgmt 0000:87:00.0: mailbox.m.9437186 ffff9ebb67bf2010 mailbox_post_response: posting response for: 10 via HW
[1879235.893672] xocl 0000:87:00.1: icap.u.23068674 ffff9ebaef434010 icap_read_from_peer: reading from peer
[1879235.893685] xocl 0000:87:00.1: mailbox.u.9437186 ffff9ebbf58f6c10 mailbox_request: sending request: 10 via HW
[1879235.896034] xclmgmt 0000:87:00.0: mailbox.m.9437186 ffff9ebb67bf2010 process_request: received request from peer: 10, passed on
[1879235.896040] xclmgmt 0000:87:00.0: xclmgmt_read_subdev_req: req kind 1
[1879235.896064] xclmgmt 0000:87:00.0: clock.m.28311552 ffff9ebbd9352010 clock_get_freq_by_id: freq = 300
[1879235.896075] xclmgmt 0000:87:00.0: clock.m.28311552 ffff9ebbd9352010 clock_get_freq_by_id: freq = 500
[1879235.896084] xclmgmt 0000:87:00.0: clock.m.28311552 ffff9ebbd9352010 clock_get_freq_by_id: freq = 450
[1879235.898084] xclmgmt 0000:87:00.0: clock.m.28311552 ffff9ebbd9352010 clock_get_freq_counter_khz: khz: 300000
[1879235.900083] xclmgmt 0000:87:00.0: clock.m.28311552 ffff9ebbd9352010 clock_get_freq_counter_khz: khz: 500000
[1879235.902082] xclmgmt 0000:87:00.0: clock.m.28311552 ffff9ebbd9352010 clock_get_freq_counter_khz: khz: 450000
[1879235.902093] xclmgmt 0000:87:00.0: mailbox.m.9437186 ffff9ebb67bf2010 mailbox_post_response: posting response for: 10 via HW
[1879235.906608] xocl 0000:87:00.1: icap.u.23068674 ffff9ebaef434010 icap_cached_ocl_frequency: no cached data for 3
[1879235.915138] xocl 0000:87:00.1: icap.u.23068674 ffff9ebaef434010 icap_cached_ocl_frequency: no cached data for 3
[1879235.967848] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 xclbin_downloaded: xclbin is already downloaded
[1879235.967855] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 xocl_read_axlf_helper: Loaded xclbin bb4375fc-b055-23c4-3754-c587632806a2
[1879235.968452] xocl 0000:87:00.1: icap.u.23068674 ffff9ebaef434010 icap_lock_bitstream: bitstream bb4375fc-b055-23c4-3754-c587632806a2 locked, ref=1
[1879235.968460] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 kds_add_context: Client pid(1868118) add context CU(0xffffffff) shared(true)
[1879235.968470] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 kds_del_context: Client pid(1868118) del context CU(0xffffffff)
[1879235.968475] xocl 0000:87:00.1: icap.u.23068674 ffff9ebaef434010 icap_unlock_bitstream: bitstream bb4375fc-b055-23c4-3754-c587632806a2 unlocked, ref=0
[1879235.968878] xocl 0000:87:00.1: icap.u.23068674 ffff9ebaef434010 icap_lock_bitstream: bitstream bb4375fc-b055-23c4-3754-c587632806a2 locked, ref=1
[... cut ...]
[1902206.605977] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 kds_del_context: Client pid(1961093) del context CU(0x0)
[1902206.605991] xocl 0000:87:00.1: icap.u.23068674 ffff9ebaef434010 icap_unlock_bitstream: bitstream bacec39b-219a-f153-fb78-e6871523cc6e unlocked, ref=0
[1902206.605999] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 xocl_destroy_client: client exits pid(1961093)
[1902206.606004] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 xocl_drvinst_close: CLOSE 2
[1902206.606008] xocl 0000:87:00.1:  ffff9ebbf6ae90b0 xocl_drvinst_close: NOTIFY 0000000073a5a7df
```
Best.

vishnuchebrolu, 2021-12-08:
Hi @IskandarZhang,
Can you please share the g++ command that was used to generate the host executable.

Thanks,
Vishnu

heeran-xilinx, 2021-12-20:
closing this as no respond for last query. Please re-open if still not yet resolve.

closed 2021-12-20 by heeran-xilinx