Xilinx/Vitis_Accel_Examples #54: System crashed when running P2P_bandwidth on U250

Evane5cence, 2021-12-01:
Hi, I used command `make all TARGET=hw PLATFORM=xilinx_u250_gen3x16_xdma_3_1_202020_1` to obtain the bandwidth.xclbin and p2p_bandwidth. However, the system crashed when running `./p2p_bandwidth -x ./build_dir.hw.xilinx_u250_gen3x16_xdma_3_1_202020_1/bandwidth.xclbin -f ./data/sample.txt`. 

Other info may be useful: I can successfully run the p2p_simple example, the p2p feature is enabled. When running p2p_bandwidth, the xclbin file was successfully programmed. The system crashed after output `Start P2P Write of various buffer sizes from device buffers to SSD`. Vitis and XRT version: 2020.2. XRT build version: 2.8.743. OS: ubuntu 18.04.

Looking forward to your reply. Thanks.

vishnuchebrolu, 2022-01-18:
@Evane5cence Can you try p2p_bandwidth design targeting u250 platform using latest Vitis and XRT Version(2021.1/2021.2). This p2p_bandwidth design initially only supported to u2 platform, then in the latest releases the support was extended to U200/U250/.. platforms.
So can you try using latest Vitis & XRT version.

Thanks,
Vishnu

heeran-xilinx, 2022-01-28:
closing now as no respond to last comments/suggestion. Please try the suggested steps, and if still facing the same issue, re-open it.

closed 2022-01-28 by heeran-xilinx