Xilinx/Vitis_Accel_Examples #56: Examples not compiling in Vitis when copying manually

StefanThorlacius, 2022-01-21:
It seems that Vitis do not accept the examples if they are not installed via Vitis IDE.
Is it not possible to just copy the files into directory Vitis_examples ? Is some extra needed to be done?

Clone https://github.com/Xilinx/Vitis_Accel_Examples
Copy Vitis_Accel_Examples to usb stick
Copy Vitis_Accel_Examples from usb stick to target machine into the directory ~/.Xilinx/Vitis/201.2/Vitis_examples

Start Vitis
Create a project, slask2, based on Vitis_Example "Hello world"
Compile

This compilation will fail because the main program can not find xcl2.hpp

The xcl2 source files are placed in
~/.Xilinx/Vitis/201.2/Vitis_examples/Vitis_Accel_Examples/common/includes/xcl2

But this directory is not included in the "include" folder in the project, 
nor is xcl2.hpp or xcl2.cpp copied to 
~/workspace/slask2/src
~/workspace/slask2/Emulation-SW/src (This directory is empty)

Note, it is possible to compile and execute the project if the makefile system is used in a terminal window.

heeran-xilinx, 2022-01-26:
Hi @StefanThorlacius ,
Can you please try following steps to import Locally downloaded repository in Vitis IDE?

In Vitis IDE, open Window > Preferences then select Xilinx > Example Repositories or Library Repositories. Here, you should see a list of library/example repos. Either set the Location of an existing repository to the local dir, or add a new repo, and set the new repository’s location:
[image]

StefanThorlacius, 2022-01-26:
That is a good tip, thanks (my path is currently correct)
I can see the examples and I can create projects based on the examples but when I create an project something goes wrong.

However the problem is of slightly different type, one of the libraries in the "include folder" gets messed up.

Note, I manually copied Vitis_Accel_Examples into 
/home/xxxxx/.Xilinx/Vitis/2021.2/Vitis_examples/Vitis_Accel_Examples
Should the content in the cloned directory Vitis_Accel_Examples be copied directly into /home/xxxxx/.Xilinx/vitis/2021.2/Vitis_examples?

If you for a project go to properties -> settings -> (folder "tool setting") -> gcc Host compiler (x86_64) -> includes
One of the include directories are defined as 
"${workspace_loc:${ProjName}/libs/common/includes/xcl2}

This path is expanded to a mess in the "include" folder for a project, and there is no .../libs/" under the project in ${workspace_loc:${ProjName}

Beside xcl2.hpp and xcl2.cpp is not copied to "${workspace_loc:${ProjName}/libs/common/includes/xcl2"
The are only located in ~/Xilinx/Vitis/2021.2/Vitis_examples/common/include/xcl2/
They are never copied to "${workspace_loc:${ProjName}/libs/common/includes/xcl2}"

vishnuchebrolu, 2022-01-31:
@StefanThorlacius  When I tried the steps to import locally downloaded repository things are working fine without any issue. Can you check again if something got messed up with your local directory?

Thanks,
Vishnu

vishnuchebrolu, 2022-02-01:
@StefanThorlacius  Can you provide the steps you tried to reproduce the issue.

-Vishnu

heeran-xilinx, 2022-02-07:
Closing this issue as no response to last query. Please re-open the issue if facing same problem. Please provide more information as requested.

closed 2022-02-07 by heeran-xilinx

StefanThorlacius, 2022-02-16:
The following steps was done

1. Install Xilinx developement enviroment on Computer X with no access to internet via portable SSD disk
2. Clone repository https://github.com/Xilinx/Vitis_Accel_Examples onto computer B into 
~/.Xilinx/vitis/2021.2/vitis_examples/Vitis_Accel_Examples

3. Copy ~./Xilinx/vitis/2021.2/vitis_examples/Vitis_Accel_Examples to a USB
4. Copy from USB to Computer X into 
~/.Xilinx/vitis/2021.2/vitis_examples/Vitis_Accel_Examples

5. Start Xilinx on computer X
6. Create a project 
7. Add the vitis exmapels to the IDE
7. Use vitis example vadd (hello_world)  as base to create the project
8. The IDE setups the project

Now

A) In the the "incude path" part of the project you can not see any references to 
~/.Xilinx/vitis/2021.2/vitis_examples/Vitis_Accel_Examples/common/include/xcl2
(where the xcl2.hpp is)
If you go into the projects settings you can see that the IDE try to add a path to 
'Emulation-SW/${ProjName}/libs/common/includes/xcl2 
Which is not resolvable so the reference to xcl2.h can not be found.

B)
The IDE do not copy (I assume it should) xcl2.cpp to the same directory as host.cpp so it wont be able to compile that either.

Note, if I, in a terminal window, change to ~/.Xilinx/vitis/2021.2/vitis_examples/Vitis_Accel_Examples/hello_world
and do 
make all TARGET=sw_emu PLATOFRM=xilinx_u50_gen3x16_xdma_201920_3 
Everything works fine.