Xilinx/Vitis_Accel_Examples #58: providing  --clock.defaultFreqHz on Alveo U200 2021.1 does not work / hangs

mayyxeng, 2022-03-15:
I was trying to manually set the clock frequency of an RTL (e.g., `rtl_vadd`) kernel  through the `--clock.freqHz` or `--clock.defaultFreqHz` option in Vitis for Alveo U200. Interestingly, no matter what frequency I provide the `hw_emu` and `hw` **hangs** (even if I provide the default value of 300000000).

Upon inspecting the waveforms, I noticed that the `ap_rst_n` signal given to the kernel is forever kept low (i.e., the kernel is constantly being reset).

I made the following simple change to the Makefile in `rtl_kernels/rtl_vadd/Makefile`.  

```diff
diff --git a/rtl_kernels/rtl_vadd/Makefile b/rtl_kernels/rtl_vadd/Makefile
index 0f58ecc4..3f38b266 100755
--- a/rtl_kernels/rtl_vadd/Makefile
+++ b/rtl_kernels/rtl_vadd/Makefile
@@ -56,6 +56,7 @@ XF_PROJ_ROOT = $(shell readlink -f $(COMMON_REPO))
 TARGET := hw
 HOST_ARCH := x86
 SYSROOT := 
+FREQ := 300000000
 
 include ./utils.mk
 
@@ -102,7 +103,7 @@ endif
 
 ############################## Setting up Kernel Variables ##############################
 # Kernel compiler global settings
-VPP_FLAGS += -t $(TARGET) --platform $(DEVICE) --save-temps 
+VPP_FLAGS += -t $(TARGET) --platform $(DEVICE) --save-temps  --clock.defaultFreqHz $(FREQ)
 ifneq ($(TARGET), hw)
        VPP_FLAGS += -g
 endif

```

Interestingly, that is not the case for U250, in fact managed to pass the `hw_emu` test just fine using the following commands:

```bash
make run TARGET=hw_emu DEVICE=xilinx_u250_gen3x16_xdma_3_1_202020_1 FREQ=300000000 -B
make run TARGET=hw_emu DEVICE=xilinx_u250_gen3x16_xdma_3_1_202020_1 FREQ=200000000 -B
```

If I replace the `--clock.defaultFreqHz` with `--kernel_frequency` then `hw` and `hw_emu` succeed on U200. [image] mentions that `--kernel_frequency` should only be used with legacy platforms and since I am using the latest U200 shell I was surprised by this observation. Is this expected?

System info:
```
XRT   2.11.634
Platform:  xilinx_u200_gen3x16_xdma_1_202110_1
Ubuntu 20.04
Kernel 5.4.0-104-generic
> vivado -version :
  Vivado v2021.1.1 (64-bit)
  SW Build 3286242 on Wed Jul 28 13:09:46 MDT 2021
  IP Build 3279568 on Wed Jul 28 16:48:48 MDT 2021
> vitis -version :
****** Xilinx Vitis Development Environment
****** Vitis v2021.1.1 (64-bit)
  **** SW Build 3275943 on 2021-07-20-22:15:00

Vitis Examples:
On branch 2021.1
commit 6b247bee4e5b4122184254d5424ede9c996c8ced
```

virata-xilinx, 2022-04-05:
Hi @mayyxeng , Your observation, and query is not specific to examples. Can you please post your query on the Xilinx Forum -
https://support.xilinx.com/s/topic/0TO2E000000YKYAWA4/vitis-acceleration-acceleration?language=en_US

Thanks
Virat

mayyxeng, 2022-04-05:
Hi Virat,

I copied my issue to the [image]:
https://support.xilinx.com/s/question/0D52E000076c0n3SAA/providing-vitis-with-clockdefaultfreqhz-on-alveo-u200-20211-does-not-work-change-the-xclbin-clock-settings-and-causes-runtime-hangs

heeran-xilinx, 2022-04-05:
Thanks @mayyxeng . We are closing this git issue as it is already getting discussed in Xilinx Forum with Vitis experts.

closed 2022-04-05 by heeran-xilinx