Xilinx/Vitis_Accel_Examples #60: Load multiples xclbin files in host code

rubende, 2022-04-26:
Hello,

I am trying to make a host code using OpenCL that uses multiple xclbin files sequentially. That is, clarifying the step by step process:

load a first xclbin
perform an operation with the kernel it contains and keep the result in the DDR4 memory of the FPGA
unload the first xclbin and load a different second one
use the result of the first kernel as input of the second one
The problem I am encountering is that, when I try to load the second xclbin, I get the following error: "ERROR: device programmed with different xclbin".

I am not finding (if it exists, please tell me which one it is) an example in this repository that does something similar to what I am asking.

What is the cause of the error? What would be the correct method to do what I propose? Putting both kernels in the same xclbin is not possible in this case due to lack of URAM.

Thank you.

vishnuchebrolu, 2022-04-26:
Hi @rubende,
I guess you can use kernel_swap design as reference.
https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/sys_opt/kernel_swap

This example shows how host can swap the kernels and share same buffer between two kernels which are exist in separate binary containers. Dynamic platforms does not persist the buffer data so host has to migrate data from device to host memory before swapping the next kernel. After kernel swap, host has to migrate the buffer back to device.

Thanks,
Vishnu

vishnuchebrolu, 2022-04-26:
Hi @rubende ,
I would suggest you to post your query on the Xilinx Forum -
https://support.xilinx.com/s/topic/0TO2E000000YKYAWA4/vitis-acceleration-acceleration?language=en_US

Thanks,
Vishnu

rubende, 2022-04-26:
Solve.

I had to free all the buffers of the first kernel before preparing the second Program.

Thank you.

closed 2022-04-26 by rubende