Xilinx/Vitis_Accel_Examples #61: Running p2p bandwidth example fails

teotro, 2022-05-16:
Hello,

I am trying to run the p2p bandwidth example.
I gave the command: **_./p2p_bandwidth -x ./bandwidth.xclbin -p /dev/nvme0n1_**, but it fails and gives the following output:

_**Trying to program device[0]: xilinx_u2_gen3x4_xdma_gc_base_2
XRT build version: 2.11.634
Build hash: 5ad5998d67080f00bca5bf15b3838cf35e0a7b26
Build date: 2021-06-08 22:10:49
Git branch: 2021.1
PID: 26427
UID: 1021
[Mon May 16 22:31:30 2022 GMT]
HOST: caslab-cloudfpga
EXE: /home/theodoros/Downloads/Vitis_Accel_Examples/host/p2p_bandwidth/p2p_bandwidth
[XRT] ERROR: See dmesg log for details. err=-2
[XRT] ERROR: failed to load xclbin: Invalid argument
Failed to program device[0] with xclbin file!**_

After a little investigation in the Internet, I found an opinion that it might be that the shell partition seems not to be programmed.
Any thoughts for troubleshooting the issue would be really helpful.

Thanks,
Theodoros

vishnuchebrolu, 2022-05-18:
Hi @teotro ,
I would suggest you to post your query on the Xilinx Forum -
https://support.xilinx.com/s/topic/0TO2E000000YKYAWA4/vitis-acceleration-acceleration?language=en_US

Thanks,
Vishnu

heeran-xilinx, 2022-07-12:
closing as issue is requested to move into forum.

closed 2022-07-12 by heeran-xilinx