Xilinx/Vitis_Accel_Examples #62: common/includes/xcl2/xcl2.cpp:46 Error calling err = cl::Platform::get(&platforms), error code is: -1001

crizy, 2022-05-18:
i run hello_world example,when make check TARGET=hw, The following error occurred

[XRT] WARNING: No devices found
[XRT] ERROR: No devices found
[XRT] ERROR: No devices found
[XRT] ERROR: No devices found
..//common/includes/xcl2/xcl2.cpp:46 Error calling err = cl::Platform::get(&platforms), error code is: -1001
Makefile:124: recipe for target 'check' failed
make: *** [check] Error 1

make check TARGET=sw_emu is ok

vishnuchebrolu, 2022-05-18:
@crizy  We don't have "check" target. Can you try running your design with "make run TARGET=sw_emu PLATFORM=xpfm>"

Thanks,
Vishnu

crizy, 2022-05-18:
> @crizy We don't have "check" target. Can you try running your design with "make run TARGET=sw_emu PLATFORM=xpfm>"
> 
> Thanks, Vishnu

@vishnuchebrolu 
   i do is as follows,
            (1)make all TARGET=sw_emu PLATFORM=xpfm
               this is ok
            (2)make check/test TARGET=sw_emu PLATFORM=xpfm
               this is ok,print message as follows,
                           Device[0]: program successful!
                           TEST PASSED

crizy, 2022-05-18:
> ERROR: Card index 0 is out of range

@@vishnuchebrolu,
     
        the error information  ERROR: Card index 0 is out of range  and   ERROR: No devices found ..//common/includes/xcl2/xcl2.cpp:46 Error calling err = cl::Platform::get(&platforms), error code is: -1001

       It seems to be the problem of pyopencl installation,but  i follows operate  https://support.xilinx.com/s/article/73055?language=en_US

      but t he above error message still appears,my pip list as follows

     pip list
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
WARNING: The directory '/home/ubuntu/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Package       Version
------------- --------
appdirs       1.4.4
asn1crypto    0.24.0
cryptography  2.1.4
decorator     4.4.2
defusedxml    0.5.0
enum34        1.1.6
idna          2.6
iniparse      0.4
ipaddress     1.0.17
keyring       10.6.0
keyrings.alt  3.0
Mako          1.1.6
MarkupSafe    1.1.1
meld3         1.0.2
numpy         1.16.6
pip           20.1.1
psutil        5.4.2
pybind11      2.8.1
pycairo       1.16.2
pycrypto      2.6.1
pygobject     3.26.1
pyliblzma     0.5.3
pyopencl      2019.1.2
pytools       2020.1
pyxdg         0.25
SecretStorage 2.3.1
setuptools    0.9.8
six           1.11.0
SOAPpy        0.12.22
supervisor    3.3.1
wheel         0.30.0
wstools       0.4.3
WARNING: You are using pip version 20.1.1; however, version 20.3.4 is available.
You should consider upgrading via the '/usr/bin/python -m pip install --upgrade pip' command.

closed 2022-07-12 by heeran-xilinx