Xilinx/Vitis_Accel_Examples #65: Vitis_Accel_Examples error

crizy, 2022-06-10:
i make alveo hello_word accelerator example from [Vitis_Accel_Examples/hello_world at master · Xilinx/Vitis_Accel_Examples (github.com)](https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/hello_world)

 

the sw_emu and hw_emu build and test are ok

but hw error , hw build is ok , hw test is error, error information as follows,

 

./host ./build_dir.hw/vadd.xclbin

XRT build version: 2.3.1301

Build hash: 192e706aea53163a04c574f9b3fe9ed76b6ca471

Build date: 2022-06-06 19:03:57

Git branch: HEAD

PID: 2727

UID: 0

[Fri Jun 10 15:16:37 2022]

HOST: ubuntu

EXE: /git_file/vitis_examples_2019_2/Vitis_Accel_Examples/hello_world/host

[XRT] ERROR: Library [liboclxdp.so](https://liboclxdp.so/) not found! XILINX_XRT not set

[XRT] ERROR: XILINX_XRT must be set

[XRT] ERROR: XILINX_XRT must be set

..//common/includes/xcl2/xcl2.cpp:46 Error calling err = cl::Platform::get(&platforms), error code is: -1001

 

 

above error information prompt :

(1) Library [liboclxdp.so](https://liboclxdp.so/) not found ,but i find this [liboclxdp.so](https://liboclxdp.so/) Library in path /opt/xilinx/xrt/lib

(2) xrt no set, but i set xrt in .bashrc on my ubuntu

my .bashrc setting environment variables as follows,

 

source /opt/Xilinx/Vivado/2019.2/[settings64.sh](https://settings64.sh/)

source /opt/Xilinx/Vitis/2019.2/[settings64.sh](https://settings64.sh/)

source /opt/xilinx/xrt/[setup.sh](https://setup.sh/)

export LIBRARY_PATH=/usr/lib/x86_64-linux-gnu

export PLATFORM_REPO_PATHS=/opt/xilinx/platforms

 

and When I open a terminal with ctrl+alt+t, the terminal display environment variable is activated,as follows,

 

XILINX_XRT   : /opt/xilinx/xrt

PATH      : /opt/xilinx/xrt/bin:/opt/Xilinx/Vitis/2019.2/bin:/opt/Xilinx/Vitis/2019.2/gnu/microblaze/lin/bin:/opt/Xilinx/Vitis/2019.2/gnu/arm/lin/bin:/opt/Xilinx/Vitis/2019.2/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/Vitis/2019.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/Vitis/2019.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/Vitis/2019.2/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/Vitis/2019.2/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/Vitis/2019.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/Vitis/2019.2/tps/lnx64/cmake-3.8.2/bin:/opt/Xilinx/Vivado/2019.2/bin:/opt/Xilinx/Vitis/2019.2/bin:/opt/Xilinx/Vitis/2019.2/gnu/microblaze/lin/bin:/opt/Xilinx/Vitis/2019.2/gnu/arm/lin/bin:/opt/Xilinx/Vitis/2019.2/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/Vitis/2019.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/Vitis/2019.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/Vitis/2019.2/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/Vitis/2019.2/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/Vitis/2019.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/Vitis/2019.2/tps/lnx64/cmake-3.8.2/bin:/opt/Xilinx/Vivado/2019.2/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

LD_LIBRARY_PATH : /opt/xilinx/xrt/lib:/opt/Xilinx/Vitis/2019.2/lib/lnx64.o/Ubuntu/18:/opt/Xilinx/Vitis/2019.2/lib/lnx64.o/Ubuntu:/opt/Xilinx/Vitis/2019.2/lib/lnx64.o:/opt/Xilinx/Vitis/2019.2/runtime/lib/x86_64:/opt/Xilinx/Vitis/2019.2/lib/lnx64.o/Ubuntu/18:/opt/Xilinx/Vitis/2019.2/lib/lnx64.o/Ubuntu:/opt/Xilinx/Vitis/2019.2/lib/lnx64.o:/opt/Xilinx/Vitis/2019.2/runtime/lib/x86_64

PYTHONPATH   : /opt/xilinx/xrt/python:

vishnuchebrolu, 2022-06-10:
Hi @crizy ,

I see you are using 2019.2 Vitis & XRT. If that is the case please use 2019.2 branch of the github repository instead of using master branch. The master branch points to latest Vitis & XRT(2022.1 version).

Thanks,
Vishnu

crizy, 2022-06-10:
> Hi @crizy ,
> 
> I see you are using 2019.2 Vitis & XRT. If that is the case please use 2019.2 branch of the github repository instead of using master branch. The master branch points to latest Vitis & XRT(2022.1 version).
> 
> Thanks, Vishnu

@vishnuchebrolu ,

sorry ,  my  link above is error, i use Vitis_Accel_Examples is 2019.2 branch(https://github.com/Xilinx/Vitis_Accel_Examples/tree/2019.2/hello_world)

vishnuchebrolu, 2022-07-07:
@crizy ,
Is the issue resolved?
Can you please make sure Vitis & XRT are sourced before running the design.

Thanks,
Vishnu

heeran-xilinx, 2022-07-12:
closing. Please re-open if issue still there and provide more info.

closed 2022-07-12 by heeran-xilinx