Xilinx/Vitis_Accel_Examples #70: must be enabled with the -std=c++0x or -std=gnu++0x compiler options

crizy, 2022-07-29:
when i make a vitis accel kernel ,  run C Synthesis 。i Encountered the following problems,

In file included from /opt/Xilinx/Vivado/2019.2/lnx64/tools/gcc/lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/random:34:
/opt/Xilinx/Vivado/2019.2/lnx64/tools/gcc/lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/bits/c++0x_warning.h:32:2: error: This file requires compiler and library support for the upcoming ISO C++ standard, C++0x. This support is currently experimental, and must be enabled with the -std=c++0x or -std=gnu++0x compiler options.
#error This file requires compiler and library support for the upcoming \
 
How can I solve this problem?
how can i pass parameter -std=c++0x or -std=gnu++0x compiler options  to synthesis section'cflags  in hls project settings dialog

crizy, 2022-08-02:
when i open hls project with vivado hls，i set  synthesis section'cflags  -std=c++0x  in hls project settings dialog,it's compiler ok，I want to know how to pass parameters std=c++0x or -std=gnu++0x compiler options to into synthesis section'cflags in hls project settings dialog by gcc/v++ commands on accelerate platform design,There seems to be no relevant parameters here,
g++ -Wall -g std=c++0x.....  has no effect,it not  pass parameters std=c++0x or -std=gnu++0x compiler options to into synthesis section'cflags in hls project

crizy, 2022-08-02:
/opt/Xilinx/Vivado/2019.2/lnx64/tools/gcc  version is 4.6.3 , Should this be upgraded to version 4.8.x?
Because I learned from the information that the c++11 feature is fully supported after the version of compiler gcc 4.8.x,how to upgraded /opt/Xilinx/Vivado/2019.2/lnx64/tools/gcc  version to 4.8.x?

keryell, 2022-08-02:
`2019.2` means a version of year 2019, which is very old. If you get a version `2022.1` for example, you will get a more modern tool chain, supporting C++14 and some C++17.

closed 2022-08-08 by crizy