Xilinx/Vitis_Accel_Examples #71: Vitis_Accel_Examples 2020.2 error

crizy, 2022-08-08:
my infomation as follows,

    os                  : ubuntu 18.04

    platform             : xilinx_u200_xdma_201830_2

    xrt                  : xrt_202020.2.8.743

    vitis/vivado           : 2020.2

    Vitis_Accel_Examples  : 2020.2

 

i make hello_word accl example,

1、 when make all sw_emu/hw_emu, the following warnings are reported:

       WARNING: /opt/Xilinx/Vitis/2020.2/tps/lnx64/jre9.0.4 does not exist.

    i see path : /opt/Xilinx/Vitis/2020.2/tps/lnx64/ which has jre11.0.2,no jre9.04，how is this related jre9.04

2、when make test sw_emu/hw_emu, the following warnings are reported:

      CRITICAL WARNING: [HW-EMU 08-0] Unable to find emconfig.json. Using default device "xilinx:pcie-hw-em:7v3:1.0"

      Trying to program device[0]: xilinx:pcie-hw-em:7v3:1.0

   this is not program xilinx_u200_xdma_201830_2, and the emconfig.json is under path _x.hw_emu.xilinx_u200_xdma_201830_2,

   why not  find emconfig.json

vishnuchebrolu, 2022-08-10:
Hi @crizy ,

Can you try to run the design using "run" target. 
make run target=sw_emu/hw_emu

Thanks,
Vishnu

crizy, 2022-08-11:
Hi @vishnuchebrolu

      make run target=sw_emu/hw_emu  is ok , I'm sorry that I didn't look at the makefile file carefully. I thought it was the same as the previous version， but how to solve the first problem? Although at present, it does not affect the implementation of accelerated design examples

closed 2022-08-11 by heeran-xilinx