Xilinx/Vitis_Accel_Examples #72: Curiosity of performance between host_memory_bandwidth and host_memory_bandwidth_xrt

anonymous1782, 2022-08-09:
Hi, 

I have a question about the performance between [host_memory_bandwidth](https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/performance/host_memory_bandwidth) and [host_memory_bandwidth_xrt](https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/performance/host_memory_bandwidth_xrt).

It seems that they have same kernel and similar host code, but the performance results are quite different.

For buffer size 32.00 KB example,

OpenCL host example is
```
Concurrent Read and Write Throughput = 12.1802 (GB/sec) for buffer size 32.00 KB
Read Throughput = 8.17289 (GB/sec) for buffer size 32.00 KB
Write Throughput = 8.26587 (GB/sec) for buffer size 32.00 KB
```

,while XRT host example is
```
Concurrent Read and Write Throughput = 12.4713 (GB/sec) for buffer size 32.00 KB
Read Throughput = 0.00397313 (GB/sec) for buffer size 32.00 KB
Write Throughput = 0.0040426 (GB/sec) for buffer size 32.00 KB
```

What is the main factor contributing to this difference? 
Do XRT and OpenCL affect the performance significantly?

vishnuchebrolu, 2022-08-10:
Hi @anonymous1782 ,

Can you please share the Vitis and XRT version that are being used.

Thanks,
Vishnu

anonymous1782, 2022-08-10:
Hi @vishnuchebrolu 

Hmm.. both results shown in the post are not my own. The results are from the README.md file in this repository. ([OpenCL](https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/performance/host_memory_bandwidth), [XRT](https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/performance/host_memory_bandwidth_xrt)) 

By the way, I'm using vitis v2021.2 and xrt v 2.12.427.

Thanks,