Xilinx/Vitis_Accel_Examples #73: Problem with accessing SSD from FPGA

omidrostamabadi, 2022-08-19:
Hello,

I have a U.2 SmartSSD ([userguide](https://www.xilinx.com/content/dam/xilinx/support/documents/boards_and_kits/accelerator-cards/1_3/ug1382-smartssd-csd.pdf)).

Is there a way to access the SSD (read/write) in the FPGA kernel? I am aware of P2P buffers, but as far as I know, in order to copy data from SSD to P2P buffer in FPGA DDR memory, host needs to initiate the transfer. But in my use case, I need to communicate with SSD in the FPGA kernel, without any involvement of the host CPU.

Any help would be appreciated.

heeran-xilinx, 2022-09-03:
Hi @omidrostamabadi ,
can you please post such query to Xilinx Forum?

https://support.xilinx.com/s/topic/0TO2E000000YKYAWA4/vitis-acceleration-acceleration?language=en_US

closed 2022-09-03 by heeran-xilinx

liu-xueyang, 2022-09-16:
The examples under /host/p2p_simple, /host/p2p_bandwidthand /host/p2p_overlap_bandwidth utilize the P2P connection between the FPGA and SSD and the source code includes the required API. The tests also need to run with root access