Xilinx/Vitis_Accel_Examples #81: 'image2d' is not supported by the current compiler

weinholtendian, 2022-09-19:
I'm having trouble building an OpenCL kernel that uses image2d and image3d. The error is:

```
ERROR: [v++ 207-5509] 'image2d' is not supported by the current compiler (/home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/src/vector_addition.cl:64:47)
```

It does not seem reasonable that an OpenCL compiler would lack support for image2d and image3d. What has happened here?

To reproduce, add this code to `ocl_kernels/cl_helloworld/vector_addition.cl`: 

```opencl
__constant sampler_t smp =
  CLK_NORMALIZED_COORDS_FALSE |
  CLK_FILTER_LINEAR |
  CLK_ADDRESS_CLAMP_TO_EDGE;

__kernel void testImage(__read_only image2d_t sourceImage,
			__global float *outputBuffer) {

  float4 v = read_imagef(sourceImage, smp,
			 (float2)((float)(get_global_id(0)),
				  (float)(get_global_id(1))));

  outputBuffer[get_global_size(0)] = v.x;
}
```

Build log:

```
~/Vitis_Accel_Examples/ocl_kernels/cl_helloworld$ make all TARGET=sw_emu PLATFORM=k26_base_starter_kit
/bin/sh: 12: [[: not found
/bin/sh: 12: [[: not found
/bin/sh: 15: [[: not found
/bin/sh: 18: [[: not found
mkdir -p ./_x.sw_emu.k26_base_starter_kit
v++ -g -t sw_emu --platform k26_base_starter_kit --save-temps  -c -k vector_add --temp_dir ./_x.sw_emu.k26_base_starter_kit  -I'src' -o'_x.sw_emu.k26_base_starter_kit/vector_add.xo' 'src/vector_addition.cl'
Option Map File Used: '/tools/Xilinx/Vitis/2022.1/data/vitis/vpp/optMap.xml'

****** v++ v2022.1.2 (64-bit)
  **** SW Build 3602371 on 2022-08-02-23:13:43
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ compile can be found at:
	Reports: /home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/_x.sw_emu.k26_base_starter_kit/reports/vector_add
	Log files: /home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/_x.sw_emu.k26_base_starter_kit/logs/vector_add
Running Dispatch Server on port: 37397
INFO: [v++ 60-1548] Creating build summary session with primary output /home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/_x.sw_emu.k26_base_starter_kit/vector_add.xo.compile_summary, at Mon Sep 19 15:34:19 2022
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Mon Sep 19 15:34:19 2022
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/_x.sw_emu.k26_base_starter_kit/reports/vector_add/v++_compile_vector_add_guidance.html', at Mon Sep 19 15:34:20 2022
INFO: [v++ 60-895]   Target platform: /tools/Xilinx/Vitis/2022.1/platforms/xilinx_k26_base_starter_kit_202210_1/k26_base_starter_kit.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/tools/Xilinx/Vitis/2022.1/platforms/xilinx_k26_base_starter_kit_202210_1/hw/k26_base_starter_kit.xsa'
INFO: [v++ 60-585] Compiling for software emulation target
INFO: [v++ 60-423]   Target device: k26_base_starter_kit
INFO: [v++ 60-242] Creating kernel: 'vector_add'

===>The following messages were generated while  performing high-level synthesis for kernel: vector_add Log file: /home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/_x.sw_emu.k26_base_starter_kit/vector_add/vector_add/vitis_hls.log :
ERROR: [v++ 207-5509] 'image2d' is not supported by the current compiler (/home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/src/vector_addition.cl:64:47)
ERROR: [v++ 207-3945] implicit declaration of function 'read_imagef' is invalid in OpenCL (/home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/src/vector_addition.cl:67:14)
ERROR: [v++ 60-300] Failed to build kernel(ip) vector_add, see log for details: /home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/_x.sw_emu.k26_base_starter_kit/vector_add/vector_add/vitis_hls.log
ERROR: [v++ 60-773] In '/home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/_x.sw_emu.k26_base_starter_kit/vector_add/vector_add/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-5509] 'image2d' is not supported by the current compiler (/home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/src/vector_addition.cl:64:47)
ERROR: [v++ 60-773] In '/home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/_x.sw_emu.k26_base_starter_kit/vector_add/vector_add/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3945] implicit declaration of function 'read_imagef' is invalid in OpenCL (/home/xilinx/Vitis_Accel_Examples/ocl_kernels/cl_helloworld/src/vector_addition.cl:67:14)
ERROR: [v++ 60-599] Kernel compilation failed to complete
ERROR: [v++ 60-592] Failed to finish compilation
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [makefile_zynqmp.mk:110: _x.sw_emu.k26_base_starter_kit/vector_add.xo] Error 1
```

I have Vitis/Vivado/etc 2022.1.2 and Ubuntu 20.04.5 LTS.

keryell, 2022-09-19:
Images are optional in OpenCL embedded profile which is the OpenCL supported by AMD FPGA.

weinholtendian, 2022-09-19:
> Images are optional in OpenCL embedded profile which is the OpenCL supported by AMD FPGA.

Okay, I guess that I'll need to implement the equivalent image operations myself, should be doable. Thank you much for the info, @keryell!