Xilinx/Vitis_Accel_Examples #85: ”ERROR: [v++ 60-398] cf2sw failed“ when run "p2p_bandwidth" with SmartSSD U2

sdudyl, 2023-03-22:
I configured the environment according to the instructions UG1382.
Then I run the vitis_accel_examples,after passed software/hardware eumulation,I make all=hw....but failed.
I also get the same wrong result when run the helloworld example.
I would like to know what causes this?

details:
`[17:07:35] Starting bitstream generation..
[17:12:37] Creating bitmap...
[17:14:27] Writing bitstream ./level0_i_ulp_my_rm_partial.bit...
[17:14:27] Finished 6th of 6 tasks (FPGA bitstream generation). Elapsed time: 00h 06m 52s 
[17:14:27] Run vpl: Step impl: Completed
[17:14:27] Run vpl: FINISHED. Run Status: impl Complete!
INFO: [v++ 60-1441] [17:14:28] Run run_link: Step vpl: Completed
Time (s): cpu = 00:00:29 ; elapsed = 01:06:37 . Memory (MB): peak = 2064.074 ; gain = 0.000 ; free physical = 24402 ; free virtual = 122272
INFO: [v++ 60-1443] [17:14:28] Run run_link: Step rtdgen: Started
INFO: [v++ 60-1453] Command Line: rtdgen
INFO: [v++ 60-1454] Run Directory: /home/embed/DYL_CSD/Vitis_Accel_Examples-2021.2/host/p2p_overlap_bandwidth/_x.hw.xilinx_u2_gen3x4_xdma_gc_2_202110_1/link/run_link
INFO: [v++ 60-991] clock name 'blp_s_aclk_kernel_ref_clk_00' (clock ID '0') is being mapped to clock name 'DATA_CLK' in the xclbin
INFO: [v++ 60-991] clock name 'blp_s_aclk_kernel2_ref_clk_00' (clock ID '1') is being mapped to clock name 'KERNEL_CLK' in the xclbin
INFO: [v++ 60-1230] The compiler selected the following frequencies for the runtime controllable kernel clock(s) and scalable system clock(s): Kernel (DATA) clock: blp_s_aclk_kernel_ref_clk_00 = 300, Kernel (KERNEL) clock: blp_s_aclk_kernel2_ref_clk_00 = 500
INFO: [v++ 60-1453] Command Line: cf2sw -a /home/embed/DYL_CSD/Vitis_Accel_Examples-2021.2/host/p2p_overlap_bandwidth/_x.hw.xilinx_u2_gen3x4_xdma_gc_2_202110_1/link/int/address_map.xml -sdsl /home/embed/DYL_CSD/Vitis_Accel_Examples-2021.2/host/p2p_overlap_bandwidth/_x.hw.xilinx_u2_gen3x4_xdma_gc_2_202110_1/link/int/sdsl.dat -xclbin /home/embed/DYL_CSD/Vitis_Accel_Examples-2021.2/host/p2p_overlap_bandwidth/_x.hw.xilinx_u2_gen3x4_xdma_gc_2_202110_1/link/int/xclbin_orig.xml -rtd /home/embed/DYL_CSD/Vitis_Accel_Examples-2021.2/host/p2p_overlap_bandwidth/_x.hw.xilinx_u2_gen3x4_xdma_gc_2_202110_1/link/int/copy_kernel.link.rtd -o /home/embed/DYL_CSD/Vitis_Accel_Examples-2021.2/host/p2p_overlap_bandwidth/_x.hw.xilinx_u2_gen3x4_xdma_gc_2_202110_1/link/int/copy_kernel.link.xml
ERROR: [CF2SW 83-2179] Unable to resolve addressing information on axi_interconnect_0, sptag bank0.  Either address assignment failed, or the PFM.AXI_PORT property on axi_interconnect_0 indicates a "memory" option that was set to an slave component and/or slave segment that are not reachable from axi_interconnect_0.  Currently instance = ddrmem_1 and segment = C0_DDR4_ADDRESS_BLOCK.  Please correct the platform design.
ERROR: [v++ 60-398] cf2sw failed
INFO: [v++ 60-1652] Cf2sw returned exit code: 255
INFO: [v++ 60-1442] [17:14:30] Run run_link: Step rtdgen: Failed
Time (s): cpu = 00:00:02 ; elapsed = 00:00:02 . Memory (MB): peak = 2064.074 ; gain = 0.000 ; free physical = 24401 ; free virtual = 122272
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.`

heeran-xilinx, 2023-04-12:
Hi @sdudyl ,
Can you please post your query to forum to get help from support team?
https://support.xilinx.com/s/topic/0TO2E000000YKYAWA4/vitis-acceleration-acceleration?language=en_US

sdudyl, 2023-04-14:
> Hi @sdudyl , Can you please post your query to forum to get help from support team? https://support.xilinx.com/s/topic/0TO2E000000YKYAWA4/vitis-acceleration-acceleration?language=en_US
Thank you!I have solved this problem.

closed 2023-04-14 by sdudyl